microsoft/AirSim #4289: About Lidar's accuracy

azadhamid, 2022-01-13:
## Question
### What's your question?
Does anyone know the accuracy of Lidar in Airsim?
I put the number of channels equal to 1,and limited the Lower and Upper level of Lidar's FOV in both directions to -1 and +1,respectively. By flying the drone in "AirSimNH" environment near a house, I collected the Lidar data as the drone moved toward the building. Unfortunately, the recorded "range to the building" was continuously around 15 although the drone approached the building (it was expected that the range value decreases). I don't know the reason, can anyone help me?

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

thomasTUK, 2022-03-11:
Did you solve the issue?

azadhamid, 2022-03-11:
> Did you solve the issue?

azadhamid, 2022-03-11:
> > Did you solve the issue?

Unfortunately, not yet.

zhong-xing-xing, 2022-05-10:
Hi!
I have same problem while flying the Drone
and I solve this problem by adding distance sensor in settings.json:
				"Distance": {
					"SensorType": 5,
					"Enabled" : true,
					"DrawDebugPoints": true,
					"MaxDistance": 200,
					"Pitch": 0.0, "Roll": 0.0, "Yaw": 90,
					"X": 0.00, "Y": 0.00, "Z": -1.00
				}
although i don't use distance sensor in my code, but the lidar works, it puts date correctly 
i don't know why ,but it works

zhong-xing-xing, 2022-05-10:
Hi!
I have same  problem while flying the Drone
you can use distance sensor to instead lidar, the distance sensor work well