microsoft/AirSim #4292: AirSim WSL2/WSLg and ROS2 not working

blakermchale, 2022-01-14:
## Bug report 

- AirSim Version/#commit: 0becf9148d88c19c3ffd69359ebe1f9c3ae43826
- UE/Unity version: 4.27
- autopilot version: NA
- OS Version: Windows 11 and WSL2 Ubuntu 20.04

### What's the issue you encountered?

I'm trying to run the ros2 wrapper like the ros1 wrapper from WSL2 with default quadrotor and call:
```
ros2 launch airsim_ros_pkgs airsim_node.launch.py host:=$WSL_HOST_IP
```
It ends up failing because of a timeout with 'getServerVersion' seen below:
<details>
<summary>
Console Output
</summary>

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [26725]
[INFO] [airsim_node-2]: process started with pid [26727]
[static_transform_publisher-1] [INFO][main():98]: Spinning until killed publishing transform from 'world_ned' to 'world_enu'
[airsim_node-2] Waiting for connection - X
[airsim_node-2] Connected!
[airsim_node-2] terminate called after throwing an instance of 'rpc::timeout'
[airsim_node-2]   what():  rpc::timeout: Timeout of 60000ms while calling RPC function 'getServerVersion'
```
</details>
It is connecting to AirSim and I'm not sure what else could be done. I've looked through the PR #3976 and it looks like the tests were done in base Ubuntu.

### Settings

Used default.

### How can the issue be reproduced?

1. Build ROS2 wrapper in WSL2 Ubuntu 20.04 with ros2 galactic
2. Run `ros2 launch airsim_ros_pkgs airsim_node.launch.py host:=$WSL_HOST_IP`

### Include full error message in text form
Seen above.

**What's better than filing an issue? Filing a pull request :).**

blakermchale, 2022-01-17:
Found the solution, there were some network settings set incorrectly.

closed 2022-01-17 by blakermchale

GateraGael, 2023-09-22:
Hello, what was the issue you were having with the settings.json file. Also were you able to actually build using x64 vs code and run the blocks environment in Unreal ?

sandilyasg, 2023-11-18:
@GateraGael  @blakermchale  where is settings.json located for this WSL setup? Is it in a Windows wsl directory or in the Documents/Airsim folder on your WSL? I am confused because I am trying to run an Airsim simulation in Unity natively in Windows but I am trying to use WSL with ubuntu 18.04 to run the ROS wrapper

GateraGael, 2024-03-06:
Hello, the settings.json file should be located on your windows 11 (c drive) Documents/Airsim. This post was back in November, have you had any luck getting the Unity version to work ?

sandilyasg, 2024-03-09:
@GateraGael thanks, yes I got the Unity version to work