microsoft/AirSim #4293: incorrect position by moveToPositionAsync

Cming7, 2022-01-17:
## Bug report 

- AirSim Version:1.2
- UE version:4.27
- OS Version:Windows 10

### What's the issue you encountered?
I encountered a similar problem with #3106 . When I tried moveToPositionAsync API. I found that the drone doesn't move if  the set point relative to the current position is less, like 0.5. 
I'm still confused witn the meaning of the lookahead and adaptive_lookahead parameters after reading https://microsoft.github.io/AirSim/apis/#lookahead-and-adaptive_lookahead. I would appreciate it if you could explain the lookahead and adaptive_lookahead parameters to me. Can I change the minimum value for lookahead?

```
client.moveToZAsync(-2, 1).join()
client.moveToPositionAsync(0.5, 0, -2, 0.1).join()
```

ppueyor, 2022-01-17:
Hi @Cming7 ,

The simpleFlight API is configured with a distance accuracy of 0.5m. If the points do not exceed that distance, the drone is set to keep stable.

>  virtual float getDistanceAccuracy() const override
> {
>             return 0.5f; //measured in simulator by firing commands "MoveToLocation -x 0 -y 0" multiple times and looking at distance traveled
>  }

You can change this DistanceAccuracy parameter in the file: AirSim/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightApi.hpp, line 197.

Hope it helps!

Cming7, 2022-01-17:
> Hi @Cming7 ,
>
> The simpleFlight API is configured with a distance accuracy of 0.5m. If the points do not exceed that distance, the drone is set to keep stable.
>
>
> You can change this DistanceAccuracy [...]

>Thank you for replying!
> I have changed the DistanceAccuracy parameter to 0.1. It improves position accuracy. For example, when I drive the multirotor forward by 1 meters in the code, the multirotor will move forward by 1.12 meters before, now it will move forward by 0.98 meters.
>But when I drive the multirotor forward by 0.5 meters or less, it still doesn't move.
>Is it effected by following code?
```
 //if adaptive lookahead is enabled the calculate lookahead error (see above fig)
                if (adaptive_lookahead) {
                    const Vector3r& actual_on_goal = goal_normalized * goal_dist;
                    float error = (actual_vect - actual_on_goal).norm() * adaptive_lookahead;
                    if (error > lookahead_error) {
                        lookahead_error_increasing++;
                        //TODO: below should be lower than 1E3 and configurable
                        //but lower values like 100 doesn't work for simple_flight + ScalableClock
                        if (lookahead_error_increasing > 1E5) {
                            throw std::runtime_error("lookahead error is continually increasing so we do not have safe control, aborting moveOnPath operation");
                        }
                    }
```
> AirSim/AirLib/src/vehicles/multirotor/api/MultirotorApiBase.cpp, line 405.

> Looking forward to your reply! Thanks a lot!

xin-peng, 2023-06-03:
Did you solve the problem? Thank you!

AlexanderZ3, 2023-07-24:
how can we adjust it in python api

turtlemicro, 2025-08-26:
did someone manage to find a solution for the python api without changing the source code?