microsoft/AirSim #4295: car_controls.steering = 0.8 means the 0.8 multiple of the max turning degree of car's tire?

dellhuyongcai, 2022-01-17:
I want to get the information of the car's tire in AirSim,recently I just know the max turning degree by the SuvFrontWheel data,it is 50 degree.When I want to make the car turn right by setting  car_controls.steering=0.8,does it means turning degree is 50*0.8=40?the turning degree keeps 40 degree while the car turns right?any help or tips will be grateful !

rhalaly, 2022-01-20:
@dellhuyongcai I also tested it using the Unreal Engine to be sure a few days ago. It seems to be true. 
The steering is the fraction of the max degree. Steering 0.8 is 50*0.8=40 degrees.

You can also test it using Unreal Engine. Change the max degree to 90 degrees for example (for better visibility) and try different steering angles. With 90 degrees and 0.5 of steering, the wheels will turn in 45 degrees.

dellhuyongcai, 2022-01-21:
Yeah.it is right,thanks for your replay.