microsoft/AirSim #4299: Segment Fault in Linux

rhalaly, 2022-01-18:
## Bug report 

- AirSim Version/#commit: 1.6.0
- UE/Unity version:
- autopilot version:
- OS Version: Ubuntu 18.04

### What's the issue you encountered?
Cannot start the 1.6.0 Blocks environment in Linux Ubuntu (Azure NV6 machine with DSVM image. 

### Settings
Not relevant

### How can the issue be reproduced?

1. Create an Azure VM of Size NV6 with Install the DSVM (Data Science Virtual Machine) image.
2. Download the 1.6.0 version of Blocks for Linux from this repository releases tab
3. Extract the zip and run it

### Include full error message in text form
<details>
  <summary>Logs</summary>

```
Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.038301
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +0:00, Platform Override: ''
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin GeometryMode
LogPluginManager: Mounting plugin EditorScriptingUtilities
LogPluginManager: Mounting plugin MeshPainting
LogPluginManager: Mounting plugin ScreenshotTools
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin MovieRenderPipeline
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin TemplateSequence
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin LauncherChunkInstaller
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin WebMMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin SignificanceManager
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin EditableMesh
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin WebMMoviePlayer
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin SoundFields
LogPluginManager: Mounting plugin GooglePAD
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AssetTags
LogPluginManager: Mounting plugin LocationServicesBPLibrary
[... 674 lines cut ...]

0x000000000371d86e Blocks!FGenericPlatformMisc::RaiseException(unsigned int)()
0x00000000038b0b97 Blocks!FOutputDevice::LogfImpl(char16_t const*, ...)()
0x000000000551fe0b Blocks!VulkanRHI::VerifyVulkanResult(VkResult, char const*, char const*, unsigned int)()
0x00000000054f153c Blocks!FVulkanSwapChain::FVulkanSwapChain(VkInstance_T*, FVulkanDevice&, void*, EPixelFormat&, unsigned int, unsigned int, bool, unsigned int*, TArray<VkImage_T*, TSizedDefaultAllocator<32> >&, signed char, FVulkanSwapChainRecreateInfo*)()
0x00000000055217d8 Blocks!FVulkanViewport::CreateSwapchain(FVulkanSwapChainRecreateInfo*)()
0x00000000055215ec Blocks!FVulkanViewport::FVulkanViewport(FVulkanDynamicRHI*, FVulkanDevice*, void*, unsigned int, unsigned int, bool, EPixelFormat)()
0x0000000005526948 Blocks!FVulkanDynamicRHI::RHICreateViewport(void*, unsigned int, unsigned int, bool, EPixelFormat)()
0x00000000059c765d Blocks!FSlateRHIRenderer::CreateViewport(TSharedRef<SWindow, (ESPMode)0>)()
0x00000000059c8309 Blocks!FSlateRHIRenderer::UpdateFullscreenState(TSharedRef<SWindow, (ESPMode)0>, unsigned int, unsigned int)()
0x0000000003ed6862 Blocks!SWindow::SetWindowMode(EWindowMode::Type)()
0x0000000006406169 Blocks!UGameEngine::CreateGameWindow()()
0x0000000005f13a8f Blocks!FDefaultGameMoviePlayer::Initialize(FSlateRenderer&, TSharedPtr<SWindow, (ESPMode)0>)()
0x0000000002c6d012 Blocks!FEngineLoop::PreInitPostStartupScreen(char16_t const*)()
0x0000000002c633aa Blocks!GuardedMain(char16_t const*)()
0x000000000750be5b Blocks!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)())()
0x00007f301623bbf7 libc.so.6!__libc_start_main(+0xe6)
0x0000000002c57029 Blocks!_start()

[2022.01.18-22.05.39:844][  0]LogExit: Executing StaticShutdownAfterError
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

```  

</details>

zimmy87, 2022-02-02:
It looks like your VM is missing Vulkan drivers. [This page](https://linuxconfig.org/install-and-test-vulkan-on-linux) should help point you in the right direction.

rhalaly, 2022-03-25:
Sorry for the long delay @zimmy87. I tried to follow the tutorial for Ubuntu Nvidia on Azure VM (NV12s_v3) and got: 
[image]

Can you please provide an example in your docs how to run AirSim on Azure Linux machines?