microsoft/AirSim #4302: Depth image frame rate slow

heleidsn, 2022-01-19:
## Question
### What's your question?

I am using the following API to get depth image:
```python
responses = self.client.simGetImages([
            airsim.ImageRequest("0", airsim.ImageType.DepthVis, Ture, True)
            ])
depth_img = airsim.list_to_2d_float_array(responses[0].image_data_float, responses[0].width, responses[0].height)
```
This works well when the depth image is small, such as 100*80, I can read depth image at about 30 FPS. 

However, when I try to increase the size of the depth image to 400*300, the frame rate decrease to 1FPS. This is too slow for some applications. 

### Include context on what you are trying to achieve

I want to get depth images at a higher FPS. 

#### Context details

### Include details of what you already did to find answers

It may because of `pixels_as_float`. I tried to set `pixels_as_float=False`, then I can get good FPS (40FPS @ 400*300 and 60 FPS @ 100*80), but I cannot get depth image. If `compress=False`, I cannot get a depth image with the correct size. If `compress=True`, the depth image will only have 0 or 255. Following is the code and depth image I get when `pixels_as_float=False` and `compress=False`.
```python
responses = self.client.simGetImages([
            airsim.ImageRequest("0", airsim.ImageType.DepthVis, False, False)
            ])
img1d = np.fromstring(responses[0].image_data_uint8, dtype=np.uint8) 
img_depth = img1d.reshape(responses[0].height, responses[0].width, 3)
```
[image]
[image]

Is there any way to get the depth image in a quick way? Or how to get the correct depth image if `pixels_as_float=False` and `compress=False`? Thanks.

xxEoD2242, 2022-01-19:
> ## Question
>
> ### What's your question?
>
>
>
> I am using the following API to get depth image:
>
> ```python
>
> responses = self.client.simGetImages([
>
>             airsim.ImageRequest("0", [...]

The best way is to turn off the UE rendering by setting DisplayMode to "NoDisplay". That usually boosts it pretty far but Python is slow in this regard. Our best bet has been to use ROS and C++ to process the images.

heleidsn, 2022-01-19:
Hi @xxEoD2242, thanks for your really quick reply!. I just tried NoDisplay, it works but can only speed up from 30Hz to 40Hz @ 100*80. I will try ROS later. This issue will be closed. Many thanks.

closed 2022-01-19 by heleidsn

xxEoD2242, 2022-01-19:
@heleidsn No worries!

Just as quick note, this can also depend heavily on the GPU that you have. We use an NVIDIA RTX 3070 and can get 30 fps at 640 x 480. You should definitely be able to get at least 15 using the ROS wrapper. If you you have questions, let me know, as I've done some work to modify a wrapper that uses a single thread to get Depth and that has worked very well for us.

heleidsn, 2022-01-19:
> 

Great! I will do some more tests and come back if still have problems. Thanks.