microsoft/AirSim #4303: z - value of IMU linear acceleration is incorrect

sudo-hello-world-2, 2022-01-19:
## Bug report 

- AirSim Version/#commit: v1.7.0-linux / #0becf9148d88c19c3ffd69359ebe1f9c3ae43826
- UE/Unity version: 4.25
- autopilot version: N/A
- OS Version: Ubuntu 20.04.2 LTS

### What's the issue you encountered?
I am implementing and Extended Kalman Filter for velocity prediction. When testing the prediction step,  accurate results for the x and y velocity are achieved. However there is an offset in the z-velocity.

 The error seems to come from an incorrect z-value of the IMU linear acceleration. I have checked the code in AirSim/AirLib/include/sensors/imu/ImuSimple.hpp and found:
https://github.com/microsoft/AirSim/blob/0becf9148d88c19c3ffd69359ebe1f9c3ae43826/AirLib/include/sensors/imu/ImuSimple.hpp#L58
I am aware of #3801 and  #3821. Apparently PX4 uses a Z-direction pointing up. Lets call this coordinate System North East Up (NEU). To my knowledge AirSim uses  North East Down (NED). So ground_truth.kinematics->accelerations.linear will also be in NED. 

When calculating the output.linear_acceleration, it is needed to convert the ground_truth.kinematics->accelerations.linear to NEU. This can be done by changing the sign of the z-value of ground_truth.kinematics->accelerations.linear. 

#### Why is the current version wrong?
Assume the Drone flies away from the ground with an acceleration of 1 m/s^2. This represents an acceleration of 

- **+**   1 m/s^2 in NEU
- **-**   1 m/s^2 in NED

in NEU the gravity is subtracted, while in NED the gravity is added:

-  **+**   1 m/s^2  **-** 9.81 m/s^2   = 8.81 m/s^2 in NEU 
-  **-**   1 m/s^2  **+** 9.81 m/s^2   = - 8.81 m/s^2 in NED

The code mentioned above gives the ground_truth.kinematics->accelerations.linear in NED, while the gravity is subtracted. This results in:

- **-**   1 m/s^2  **-** 9.81 m/s^2   = - 10.81 m/s^2 in "NEU"

As you can see the result is neither in NED nor in NEU.

To Fix this, the sign of the z-value of the ground_truth.kinematics->accelerations.linear should be multiplied by -1. This would result in:

- **+(-)1***   1 m/s^2  **-** 9.81 m/s^2   = 8.81 m/s^2 in NEU 

#### Summary
At the Moment output.linear_acceleration is neither in NED nor NEU. For support with PX4 it should be in NEU. To achieve this multiply the **z**-value of ground_truth.kinematics->accelerations.linear with **-1**.
 This transforms ground_truth.kinematics->accelerations.linear from NED to NEU. 
After that output.linear_acceleration will be in NEU.

### Settings

### How can the issue be reproduced?

This bug does not result in a program error. It can be reproduced by using the given AirSim version. I also cannot provide a test other than the logic explained above. However my Extended Kalman Filter works with the fix mentioned above and there is no offset in the z-value of the velocity anymore.

### Include full error message in text form
There is no error message

**What's better than filing an issue? Filing a pull request :).**
### Side Notes
This part is for the interested reader. It is not a bug.
Rotating output.linear_acceleration from NEU (after the fix) to body coordinates as done in 
https://github.com/microsoft/AirSim/blob/0becf9148d88c19c3ffd69359ebe1f9c3ae43826/AirLib/include/sensors/imu/ImuSimple.hpp#L62-L64

will produce other results, than rotation from NED to body coordinates. This might result in differences when comparing data of other sensors in body coordinates with the accelerometer in "body coordinates". You will need to transform both to NED first to get comparable results.

cindrew, 2022-02-07:
yes i agree, i test it with airsim in the UE and see the data with API getImuData(), it indeed should change the signs of gravity from - to + 

[AirSim/AirLib/include/sensors/imu/ImuSimple.hpp]

Line 58 in [0becf91](https://github.com/microsoft/AirSim/commit/0becf9148d88c19c3ffd69359ebe1f9c3ae43826).

also ,problem#3801 get the same

MinnDevelopment, 2022-02-17:
I'm running into the same problem, are there any plans to update this?

zimmy87, 2022-03-07:
We've discussed this matter internally at length and have come to the conclusion that our current implementation is by design. It may seem counterintuitive, but gravity being subtracted is done to counteract gravity, not to represent it. Basically, if you imagine a stationary IMU, the reading of -9.8 m/s^2 in the Z-axis indicates the IMU is being "accelerated" 9.8 m/s^2 upwards, an amount proportional to the acceleration due to gravity, resulting in the object staying stationary. Another way to think about this is to think of the -9.8 m/s^2 as the acceleration due to the normal/contact force of the ground. 

[This article](https://embeddedinventor.com/what-is-an-imu-sensor-a-complete-guide-for-beginners/) states that an IMU's internal operation can be conceptualized as a box containing a mass suspended between two springs, where the position of the mass determines the output of the IMU. When a box like this is sitting on a table, the mass pushes/pulls down on its springs, just like the mass would also push/pull down on its springs due to an upward acceleration (because of inertia), so the IMU reads both as acting in the -Z direction. For a free-falling box, the mass will gravitate towards the center where there is no push/pull on the springs, and the IMU will eventually report 0 acceleration, reflecting how it is no longer affected by the normal/contact force of the ground.

To see how real-life IMUs handle gravity, take a look at [this video](https://www.youtube.com/watch?v=MAR4yvxQpzc) around the 2:52 mark. You can see in this video that the IMU is reporting an acceleration of -G in the Z-axis with +Z pointing down: IE it is reporting an upward acceleration.

Also, I cannot personally find any documentation stating that PX4 uses a NEU coordinate system (it very well could be, but I cannot confirm this, so if anybody knows a place where this is explicitly stated, please let me know, I'd love to have more specification here). Regardless, the IMU readings are sent to PX4 via the [HIL_SENSOR mavlink message](https://mavlink.io/en/messages/common.html#HIL_SENSOR) which is documented as being in NED, so there isn't a need to convert to a different coordinate system here.

closed 2022-03-07 by zimmy87

sudo-hello-world-2, 2022-03-07:
Hi @Zimmy87,
Thank you for your reply! I agree with the explanation presented by you and your colleagues. The error was totally on my side. I have made some tests with my built-in smartphone IMU and the results are consistent with the expected. Sorry for the trouble.