microsoft/AirSim #4304: Does multirotor support moveByAngleZAsync?

Cming7, 2022-01-20:
## Question
### What's your question?

>Does multirotor support moveByAngleZAsync? What's the TODO meaning in the 
[Unity APIs](https://microsoft.github.io/AirSim/unity_api_support/)?

### Include context on what you are trying to achieve

>I want to drive the multirotor rotate on pitch angle and roll angle. I just find the rotateToYawAsync API to drive the multirotor rorate on yaw angle.

#### Looking forward to your reply. Thank you so much!

sudo-hello-world-2, 2022-01-20:
Hey @Cming7,
I think TODO just means to do. Try using moveByAngleRatesZAsync(roll_rate, pitch_rate, yaw_rate, z, duration, vehicle_name=''). If you need to, check the implemented API page for more details (https://microsoft.github.io/AirSim/api_docs/html/ ).

Gastastrophe, 2022-01-20:
Are you building Airsim on Unity? If an API says TODO in the Unity APIs, then that means it works in Unreal Engine but not Unity.

Cming7, 2022-01-21:
> Are you building Airsim on Unity? If an API says TODO in the Unity APIs, then that means it works in Unreal Engine but not Unity.

I build the project on UE4.27.

zimmy87, 2022-02-02:
Hi @Cming7, the server binding for this API was removed in #2297. You'll want to use the `moveByRollPitchYawZAsync` API instead.

zimmy87, 2022-02-02:
I put out PR #4338 to fix this.

PR #4338 (2022-02-02): deprecate moveByAngleZAsync and moveByAngleThrottleAsync

Fixes: #4304    

## About
This PR deprecates moveByAngleZAsync and moveByAngleThrottleAsync because their server bindings don't exist anymore. These APIs were replaced by moveByRollPitchYawZ and moveByRollPitchYawrateThrottle in #2297

## How Has This Been Tested?
The test script was run calling each of the deprecated APIs to confirm the correct log message is outputted and the new APIs are called instead.

## Screenshots (if appropriate):
closed 2022-02-02 by jonyMarino