microsoft/AirSim #4309: Adding Warthog simulation feature to AirSim

akhil22, 2022-01-21:
## What feature are you suggesting?
### Overview:

I would like to add a Warthog Simulation feature to AirSim [Warthog](https://clearpathrobotics.com/warthog-unmanned-ground-vehicle-robot/). It is an excellent platform to test off-road autonomy algorithms. The focus of my research is to come up with RL based
Controllers for different off-road vehicles and Unreal Engine with AirSim is the perfect choice for simulation platform. 
### Smaller Details: 
I have already created the warthog Pawn in Unreal with a simple PID controller and now I would like to add this to AirSim.
After looking at the source code I have created a simple hierarchical diagram which captures the basic dependency(represented by black arrows), inheritance structure(Represented by thick red arrows) and the code that needs to be written (light blue oval nodes)
[image]

### Nature of Request:  

- Addition
  - 

## Why would this feature be useful?
There has been a lot of interest in off-road autonomy for ground vehicles recently and this feature can provide a good simulation 
platform for training RL based autonomy algorithms.

Gastastrophe, 2022-01-22:
My lab would also be interested in this. There aren't a lot of platforms that support ground and air vehicles. It seems like a lot of work, but certainly not a bad idea

akhil22, 2022-01-22:
Yes its a lot of work but I think its a useful features. Do you have any specific ground platform that you want to simulate ?

akhil22, 2022-01-22:
> My lab would also be interested in this. There aren't a lot of platforms that support ground and air vehicles. It seems like a lot of work, but certainly not a bad idea

Yes its a lot of work but I think its a useful feature. Do you have any specific ground platform that you want to simulate ?

jonyMarino, 2022-02-03:
Maybe [this work](http://www.mitchellspryn.com/2019/08/05/Robust-Offroad-Rover-For-Sim2Real-Applications.html) made with an AirSim fork can help

akhil22, 2022-03-17:
> My lab would also be interested in this. There aren't a lot of platforms that support ground and air vehicles. It seems like a lot of work, but certainly not a bad idea

Hi Gastastrophe,
I have added warthog simulation to AirSim please checkout this [repo](https://github.com/akhil22/AirSim/tree/warthog) if you want to use it for your work and let me know if you face any issues.