microsoft/AirSim #4310: Clock speed doesn't work, timestamp is not synchronous

Leon-Redfield, 2022-01-22:
## Bug report 

- AirSim Version: 1.4
- UE version: 4.25
- OS Version:Ubuntu 20.04

Hi! I'm using multirotor SimMode to simulate a drone flying in the village. And I'm using "Airsim_ROS_wrapper" to receive the messages from Airsim. There are two main problems:
1. I tried to modify the clock speed in settings.json, but when I modify the value from 1 to 2, for example, 1.5, 1.7 and 1.9, the real clock speed shows like this picture: the config clock speed is different from the actual speed.
[image]
[image]
[image]

You can see, whatever value between 1-2 I use, the actual clock speed is always around 0.95. But when I modify this value to 2.0, the actual clock speed jumps to 1.76 like this:
[image]

What I really want is to make the actual clock speed just at 1.0000 but I have no idea. My CPU is i7-10700, graphic card is RTX3070Ti.
So just because of the problem above, I have the second problem.
2. The timestamp of the simulation environment is not synchronous with ROS. In airsim_ros_wrapper, I use two kinds of functions to acquire the timestamp: `airsim_timestamp_to_ros(msg)`  and `ros::Time::now()`. So I print the timestamps in the terminal, and the code is as below:
`lidar_msg.header.stamp = airsim_timestamp_to_ros(lidar_data.time_stamp);`
`ROS_WARN("Compare the two kinds of timestamp");`
`std::cout << " The lidar timestamp is : " << lidar_msg.header.stamp << std::endl;`
`std::cout << " The ros timestamp is : " << ros::Time::now() << std::endl;`
And the final result is that the difference between this two kinds of timestamps is bigger and bigger. I guess it is because of the clock speed of Airsim is not synchronous with that in ROS.
[image]

So all above, I want to solve these two problems, hope to get the solutions, thank you!

Leon-Redfield, 2022-11-09:
> Hello, How did you get around this problem? @Leon-Redfield

Sorry but I didn't solve this problem.

xuqingyuan2000, 2023-08-15:
I ran into the same issue, causing the lidar timestamp to be out of sync with the Imu timestamp. Can I use all ros::Time::now()?