microsoft/AirSim #4311: Recording distance sensor data for drones in airsim_rec.txt

bineet-coderep, 2022-01-23:
## Question
### What's your question?
I am currently using a drone to gather a dataset. Can you please let me know how can I equip the drone with a distance sensor to compute its altitude. Note that the distance sensor, in my case, needs to be pointed downwards. Merely using (x,y,z) coordinates of the drone is not sufficient to compute altitude of the drone, in my case. The environment I am currently using has animated objects moving in the ground. So, to calculate the altitude of the drone, w.r.t to those objects, I need a distance sensor that's pointed downwards. Note that I also need to record this distance sensor data is `airsim_rec.txt`.

### Include context on what you are trying to achieve
Following is what I am trying to achieve:
* Equip the drone with a distance sensor that's pointed perpendicularly downwards (towards the ground).
* Record the distance sensor data in `airsim_rec.txt`.

#### Context details

Development environment:
* OS: Ubuntu 20.04.3 LTS (64 bit)
* Python: 3.7.6
* Unreal: 4.25.4

I am currently recording x,y,x and roll, pitch yaw, in `airsim_rec.txt`, by modifying `PawnSimApi.cpp` ([Issue 4251](https://github.com/microsoft/AirSim/issues/4251)) 

### Include details of what you already did to find answers
I tried the following, without success:
* [Distance senor AirSim documentation](https://microsoft.github.io/AirSim/distance_sensor/).
* [Sensors in AirSim](https://microsoft.github.io/AirSim/sensors/).
* [Issue 2444](https://github.com/microsoft/AirSim/issues/2444).

sudo-hello-world-2, 2022-01-25:
Hi @bineet-coderep,
**Question 1**: Equip the drone with a distance Sensor:
You will need to edit the Settings.json as referred in https://microsoft.github.io/AirSim/sensors/ . To add a distance sensor pointing downwards, add
`"Distance": {
          "SensorType": 5,
          "Enabled" : true,
          "MinDistance": 0.2,
          "MaxDistance": 40,
          "X": 0, "Y": 0, "Z": 0,
          "Yaw": 0, "Pitch": -90, "Roll": 0,
          "DrawDebugPoints": true
          }
`
in your Sensor list.

**Question 2**: Record the distance sensor data in airsim_rec.txt
If you are controlling your drone with simple_flight you can try the following pseudo python code:
`thread = client.moveToPositionAsync(x,y,z,velocity)
`

`while Vehicle is not close to end point:`
               `distance = client.getDistanceSensorData(distance_sensor_name="", vehicle_name="").distance`

`thread.join()`

To store the data, you could write to a file or redirect all print statements via the shell ( >airsim_rec.txt ). Another approach would be to modify recordings: https://microsoft.github.io/AirSim/modify_recording_data/

Hope this helps.

rajat2004, 2022-01-27:
https://microsoft.github.io/AirSim/modify_recording_data/ This page will  be helpful for details on how to modify the data being recorded. Instead of Imu or Barometer, you'll need to use `getDistanceSensorData()`, You can search for this method in the project and how it's used, the data format being returned

deepchokshi, 2022-05-23:
To add distance sensor to the drone is Airsim you need to add the configuration of distance sensor in settings.json. Configuration for distance sensor is mentioned below
"Distance": {
              "SensorType": 5,
              "Enabled": true,
              "MinDistance": 0.2,
              "MaxDistance": 40,
              "X": 0,
              "Y": 0,
              "Z": -1,
              "Yaw": 0,
              "Pitch": 0,
              "Roll": 0,
              "DrawDebugPoints": false
            }