microsoft/AirSim #4312: Python simGetImages throws unknown exception

mmuetert, 2022-01-24:
Hello,

I am trying to run the segmentation.py  example from the PythonClient folder in airsim. When I try to run the script, it runs fine until the line

#get segmentation image in various formats
responses = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.Segmentation, True), #depth in perspective projection
    airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False)])  #scene vision image in uncompressed RGBA array
print('Retrieved images: %d', len(responses))

where I get the error

"msgpackrpc.error.RPCError: rpclib: function 'simGetImages' (called with 3 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available."

I had the same error in a custom script I was writing so I decided to test the issue with the provided python script. I test the c++ version of HelloDrone and it runs simGetImages just fine.

I am running python3.6 on Ubuntu 20.04. My python packages are up to date and my Airsim install should be as well. I very recently upgraded from Ubuntu 18.04 to Ubuntu 20.04, but I never ran the python versions on 18.04, so I can't say if they worked before the upgrade.

Any help would be appreciated!

sudo-hello-world-2, 2022-01-25:
Hi @mmuetert.
The Code above works fine fine for me. Are you sure "0" is your camera name?

zimmy87, 2022-01-31:
This might be related to #4120, can you apply the fix in #4321 and see if that fixes the issue for you?

Hangero, 2025-01-07:
Hello, I encountered the same issue. May I ask how you resolved it?

Hangero, 2025-03-28:
> Hello, I encountered the same issue. May I ask how you resolved it?

Hello, for anyone using the C++ or Python version of the AirSim ROS package and encountering this issue, my understanding is that it is caused by version conflicts in the API. In January, I was using the C++ version of the AirSim ROS package, and after several reinstallations of AirSim, I eventually chose AirSim version 1.5.0, and the scene I used was v1.5 of AirSimSH, which resolved the issue I encountered. Today, after pulling my classmate's code, I encountered the same issue again, but this time I was using the Python version of the AirSim ROS package. I noticed that the version installed by running `pip install airsim` was 1.8.1, so I used `pip install airsim==1.5.0,` which successfully resolved the issue. This problem previously took me a week or two of effort to troubleshoot without finding the right solution. I hope this helps save time for others! If anyone has a solution for the C++ version without needing to reinstall, please feel free to reach out. Thank you!