microsoft/AirSim #4315: Build error (Ubuntu): cmake cannot find LLVM

yannbouteiller, 2022-01-25:
## Bug report 

- AirSim Version/#commit: https://github.com/microsoft/AirSim/commit/0becf9148d88c19c3ffd69359ebe1f9c3ae43826
- UE/Unity version: UE 4.25
- autopilot version: N/A
- OS Version: Ubuntu 18.04.2 LTS

### What's the issue you encountered?

When following the build instruction for Linux from [this page](https://microsoft.github.io/AirSim/build_linux/), I encounter the following error when running `./build.sh` for AirSim (after running `./setup.sh`):

```terminal
yann@yann-ub:~/Desktop/git/AirSim$ ./build.sh
+ debug=false
+ gcc=false
+ [[ 0 -gt 0 ]]
+ RPC_VERSION_FOLDER=rpclib-2.3.0
+ '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/home/yann/.local/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ false
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /home/yann/.local/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.0
-- The CXX compiler identification is Clang 8.0.0
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
CMake Error at cmake-modules/CommonSetup.cmake:61 (find_package):
  Could not find a package configuration file provided by "LLVM" with any of
  the following names:

    LLVMConfig.cmake
    llvm-config.cmake
[... 131 lines cut ...]
AirSim setup completed successfully!
************************************
```

### Settings

### How can the issue be reproduced?

### Include full error message in text form

CMake Error at cmake-modules/CommonSetup.cmake:61 (find_package):
  Could not find a package configuration file provided by "LLVM" with any of
  the following names:

    LLVMConfig.cmake
    llvm-config.cmake

  Add the installation prefix of "LLVM" to CMAKE_PREFIX_PATH or set
  "LLVM_DIR" to a directory containing one of the above files.  If "LLVM"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  rpclib_wrapper/CMakeLists.txt:6 (CommonSetup)

**What's better than filing an issue? Filing a pull request :).**

9woods123, 2022-05-26:
hello! have  you  solved it ? i come into the same with you,

9woods123, 2022-05-26:
I turn to the airsim 1.4.0 for linux, and it build successfully