microsoft/AirSim #4320: airsim can not detect accuracy dynamic objects' 2D_box

chongjeelee, 2022-01-26:
## Bug report 

- AirSim Version/#commit:v1.6
- UE/Unity version:UE4.25
- autopilot version:-
- OS Version:win10

### What's the issue you encountered?

when i set objects moving with blueprint spline,i found it can not detect the accuracy position when object moving on sky.Just like this below:

https://user-images.githubusercontent.com/30140969/151135810-152fe722-a6f2-42a2-a4b6-fed2bef351af.mp4

### Settings

[image]

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

chongjeelee, 2022-01-28:
i have solved this problem with blueprint!

closed 2022-01-28 by chongjeelee

Seeseallllll, 2023-02-21:
> i have solved this problem with blueprint!
Hello, I also encountered the generation problem of 2d bounding box, how do you use the blueprint to solve it? Could you share it in detail? Thanks!

bing-zhub, 2023-05-06:
> i have solved this problem with blueprint!

Same problem, how do you fix this problem using BluePrint? Could you give us a more detail solution?

Appreciated!