microsoft/AirSim #4337: Configuring Taranis QX7 to control the drone

zardosht, 2022-02-02:
## Question
### What's your question?
Hi. Can you please provide some instructions on how to configure Taranis QX7 for controlling the SimpleFlight drone? The config file provided in the docs (https://microsoft.github.io/AirSim/remote_control/#rc-setup-for-default-config) seems to be incompatible with our RC (maybe too old?)

### Include context on what you are trying to achieve

We are using 
* Taranis QX7 Access
* Update firmware to the latest: 2.3.14 (2021-07-23)
We copied the .bin file into SD card. When trying to load the model into the RC we get the message "Incompatible". 

#### Context details

### Include details of what you already did to find answers

jonyMarino, 2022-02-03:
Hi, and welcome @zardosht! can you use AirSim with PX4? In that way, you can connect your RC to QGroundControl.

zardosht, 2022-02-04:
Thanks. I will give the HITL with PX4 a try and write the result here. In the meantime I created a model from scratch that works.
The model needs to have channels set from 1 to 4 in this order: Throttle, Roll (aileron), Pitch (elevation), Yaw (rudder). 
This seems to be the channel order that AirSim expects. 
See for example the `motorMixer_t` strcut in `AirSim/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Mixer.hpp`, or the `"Joystick (T,R,P,Y,Buttons):"` message in `Unreal/Plugins/AirSim/Source/PawnSimApi.cpp`.  
I will share the model as soon I have figured out how to configure other things like Arm/Disarm etc.

jonyMarino, 2022-02-04:
That is excellent! You can use PX4 in SITL mode; HITL is not needed.