microsoft/AirSim #4339: simGetImages API causes a Vulcan Error for high resolution images

bozcani, 2022-02-07:
## Bug report 

- AirSim Version/#commit: 1.6.0
- UE/Unity version: 4.27.1
- autopilot version:
- OS Version: Ubuntu 20.04
- GPU: NVIDIA GeForce RTX 3060 Laptop GPU, 6GB 

### What's the issue you encountered?

I see that pixel formats in PIPCamera.cpp have been changed a while ago in this commit: https://github.com/microsoft/AirSim/commit/705a366ef4daeff8053df3f64b90786b7c9d9903

This change causes an error if `simGetImages` API is called for a relatively higher resolution image request (i.e, 2048, 1152). See the error below:

```
[2022.02.07-08.39.54:241][288]LogVulkanRHI: Error: VulkanRHI::vkDeviceWaitIdle(Device) failed, VkResult=-4
 at <path to UE4.27>/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1344 
 with error VK_ERROR_DEVICE_LOST
Fatal error: [File:<path to UE4.27>/Engine/Source/Runtime/VulkanRHI/Private/VulkanUtil.cpp] [Line: 958] 
VulkanRHI::vkDeviceWaitIdle(Device) failed, VkResult=-4
 at <path to UE4.27>/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1344 
 with error VK_ERROR_DEVICE_LOST
Segmentation fault (core dumped)
```
I think the reason is when you switch pixel format to Float, it requires more memory. There is no problem with commits before https://github.com/microsoft/AirSim/commit/705a366ef4daeff8053df3f64b90786b7c9d9903

However, after https://github.com/microsoft/AirSim/commit/705a366ef4daeff8053df3f64b90786b7c9d9903, the error appears when you request high resolution images vie `simGetImages` API.

### Settings

Related piece of Settings:
```
  "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 2048,
          "Height": 1152,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        },
        {
          "ImageType": 5,
          "Width": 2048,
          "Height": 1152,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        }
    ],
    "Gimbal": {
      "Stabilization": 1,
      "Pitch": -45, "Roll": 0
    }
  }
```
### How can the issue be reproduced?

1. Run `simGetImages` API with the above settings.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-02-21:
Thanks for the report. We will fix it.

zimmy87, 2022-02-24:
Hi @bozcani, I've put out a fix for this issue in #4372, please retest with the fix applied and let me know if you notice any additional issues.

bozcani, 2022-03-01:
It works with the fix. Thanks!

PR #4372 (2022-03-08): add setting for overriding pixel format in capture component

Fixes: #4339    

## About
This PR undoes #4321 and adds a new setting that allows the user to override the pixel format that each capture component in a camera is initialized with. This allows us to support scenarios where pixel format is expected to be either int or float. See #4339, #4120 and  #4321 for more information about various scenarios that are impacted by pixel format selection. The new settings include a "UnrealEngine" setting, which contains a "PixelFormatOverride" setting, which then contains a list of settings that h [... cut]
closed 2022-03-08 by jonyMarino