microsoft/AirSim #4344: Cannot build AirSim in Blocks Windows UE4.27

HarrySoteriou, 2022-02-08:
I was tweaking the settings file, trying to get the front rotors out of the simGetImage() field of view (when using a custom front camera the two front rotors would be part of the captured image no matter what I changed, i.e. pitch: -90.0, x : 3, z: 5--> NOTHING CHANGED WHAT MY simGetImage was returning )...

I tried using external camera= True, in my script without modifying the settings file... this has caused a crash and even when reverting the change AirSim will not load.... After trying to fix the issue without success I have tried building everything from scratch again on 2 separate computers using the same version UE=4.27. I have a Block solution (Block.sln), that matches the UE4 version, AirSim Plugins are placed in the project but I don't get prompted to use AirSim as soon as I load the environment.

Steps:
1. Download the Epic Games Launcher
2. Install Unreal Engine 4.27.2
3. Install Visual Studio 2019 with Desktop Development with C++ and Windows 10 SDK 10.0.18362 (went back and unchecked the even newer version of windows 10 SDK to ensure I am using all the exact same settings)
4. Using Developer Command Prompt for VS 2019, clone repo /cd AirSim/build.cmd
5. Navigate to AirSim\Unreal\Environments\Blocks and run update_from_git.bat
6. Open up Block.sln using options DebugGame_Editor and Win64. Hit F5

Expected behaviour:
Load in the environment and be able to Play AirSim

Observed behaviour:
Load in the environment and although the plugin is listed I do not have the option to run AirSim

HarrySoteriou, 2022-02-09:
For the love of god make a clear tutorial on how to build the blocks environment and common issues that are faced, this is the 3rd day of me trying to return to the point I was before editing my settings file.
The documentation at different point says to do different things:
1) Do I need to use the native command prompt or the developer command prompt VS2019 and should i use the latest windows10 SDK or the one stated in the docs?
2) Now with the pre-released airsim 1.7 can I just git clone the repo or do I have to download the 1.6 zip file?
3) Must i run the update_from_git.bat or GenerateProjectFiles.bat given the same reason as 2?
4) Why some settings edit break the entire thing even after reverting the change?

I dont have the option to PLAY AIRSIM, when I enter my blocks environment. I have ran the  DebugGame_Editor X64 and then build using "Develop Editor" and  x64

HarrySoteriou, 2022-02-09:
followed this tutorial: https://airsim-fork.readthedocs.io/en/latest/unreal_custenv.html and solved my issues, if airsim crashes soon after running a script containing simGetImages ensure that the settings.json file doesnt have any invalid settings

jonyMarino, 2022-02-22:
Hi @HarrySoteriou!
Thanks for keeping the issue up to date.

1_ The developer command prompt is only necessary to run build.cmd
2_ With binaries, it is not required to clone the repo
3_ update_from_git.bat or GenerateProjectFiles.bat are not called in the case of binaries
4_ You already solved it.

Sorry about the state of the documentation. You can submit a pull request to improve it.

ClarifiedfishLee, 2023-11-21:
> Manage to overcome this issue by ensuring that I was using the right version of VS when building the Blocks environment and returning my settings.json to the original one (On that computer I have 2 versions of VS: 2019 and 2022.. )
> 
> Then I ran the simulation using my configured settings.json/ not the default one and got the same error as [aliyun/aliyun-oss-cpp-sdk#108](https://github.com/aliyun/aliyun-oss-cpp-sdk/issues/108)
> 
> Edit: Resolved this issue by return settings.json to it's default value

Hello! I have a stupid question here. I'm using AirSim 1.8.1 builded in VS 2022 and the Block project is opened in 2022 by default. However, when I try to build Block project, it shows using Using Visual Studio 2019 14.29.30152 toolchain and failed. Is that also the problem of default VS version? How can I change the settings.json file cause I can't find it.
[image]
[image]