microsoft/AirSim #4345: Unable to view the Subwindow

pdx97, 2022-02-08:
Hi i am unable to view my subwindow in the Unreal engine for my drone even though the camera name in the code and the setting.json are same, i have read various issues and even followed them but nothing is working out .
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ClockSpeed": 2,
  "Vehicles": {
	  "Drone1": {
	  "VehicleType": "Simpleflight",
	  "AutoCreate": true,
	  "Sensors": {
		"LidarSensor1": {
		  "SensorType": 6,
		  "Enabled": true,
		  "NumberOfChannels": 16,
		  "PointsPerSecond": 10000,
		  "X": 0,
		  "Y": 0,
		  "Z": -1,
		  "DrawDebugPoints": false
		},
		"MyLidar2": {
		  "SensorType": 6,
		  "Enabled": true,
		  "NumberOfChannels": 4,
		  "PointsPerSecond": 10000,
		  "X": 0,
		  "Y": 0,
		  "Z": -1,
		  "DrawDebugPoints": false
		}
	  }
	}
},
"SubWindows": [
    			{"WindowID": 0, "CameraName": "0", "ImageType": 3, "VehicleName": "Drone1", "Visible": true, "External": true},
   		        {"WindowID": 1, "CameraName": "1", "ImageType": 5, "VehicleName": "Drone1", "Visible": true},
    			{"WindowID": 2, "CameraName": "1", "ImageType": 0, "VehicleName": "Drone1", "Visible":true}
  		]
	
}

rajat2004, 2022-02-10:
You have External set for the first subwindow but it's not an external camera. Try removing this param

pdx97, 2022-02-10:
@rajat2004 i tried that too still the subwindow doesn't appear.

rajat2004, 2022-02-12:
Wonder why things are even working for you, you don't have `SimMode` set in the settings, add that and it should work