microsoft/AirSim #4347: strange propeller animation

noahzn, 2022-02-10:
## Question
Strange propeller animation

I tried to replace the default drone with my own drone. I also replaced 4 propellers. But the animation is not correct.

[image]
### Include context on what you are trying to achieve
 The propellers operate correctly.

#### Context details

The settings.json file should be no problem.

### Include details of what you already did to find answers
N/A

noahzn, 2022-02-21:
Can someone help?

noahzn, 2022-02-25:
Hi @jonyMarino , I changed the model according to this video https://www.youtube.com/watch?v=Bp86WiLUC80
The 3d model is in .dae format, and I convert it to .fbx in Blender.

jonyMarino, 2022-05-02:
@noahzn sorry for the late response. You don't have the center of the blueprint in the center of the propeller.

closed 2022-11-30 by noahzn