microsoft/AirSim #4348: How to disable manual keyboard controls for car

TheFoolishPupil, 2022-02-11:
## Question
### What's your question?
I'm wondering how to disable the manual controls of the cars altogether. I am attempting to control them via the Python api and would like to use the free-cam mode to position the camera. However when positioning the camera with WASD and the arrow keys, both the cars that I am simulating respond to the keyboard commands as well. Is there a way to disable manual control entirely?

### Include context on what you are trying to achieve
N/A

#### Context details
Let me know if you feel any extra details would be relevant and I can provide them.

### Include details of what you already did to find answers

rajat2004, 2022-02-11:
Can you try setting `RC` param for each vehicle to 0 or something else, assuming that you don't have a remote controller attached.

TheFoolishPupil, 2022-02-14:
Thanks for the reply @rajat2004. I have tried setting ` "RC": { "RemoteControlID": 0 }` for each car but still see the same behaviour. I have tried with numerous values other than 0 as well.

rajat2004, 2022-02-28:
Sorry for the late reply, have you tried calling `enableApiControl` for the car before positioning the cameras? You could call this in the beginning, and then wait for keyboard input in the script before proceeding to the actual testing.

jonyMarino, 2022-06-09:
You can also comment [this line](https://github.com/microsoft/AirSim/blob/228f21c7595aee18515ffefc9599bdc77c07fdfc/Unreal/Plugins/AirSim/Source/Vehicles/Car/CarPawnSimApi.cpp#L138) if you need the keyboard disabled at the beginning.