microsoft/AirSim #4356: Not Found Skeletal Mesh Actor

ZhangFanSC, 2022-02-16:
## Question
### What's your question?

In my Content Browser，Skeletal Mesh Actor seems to be missing.

### Include context on what you are trying to achieve

I wonder where I can import from Skeletal Mesh Actor.Thanks.

#### Context details

[image]
[image]
@Jonathan @mattetti @mtodd 

### Include details of what you already did to find answers

jonyMarino, 2022-02-24:
@ZhangFanSC in that project without the AirSim plugin, can you visualize the skeletal mesh actor?

ZhangFanSC, 2022-02-25:
no，there's only one coordinate axis. 
1.I wonder if it is because I downloaded Airsim without downloading the plug-in correctly?
2.What should I do, try to add it without removing or affecting the existing AirSIM?