microsoft/AirSim #4361: Airsim SITL Copter overshoots

LightCannon, 2022-02-20:
## Question
### What's your question?
Airsim Copter Oscillations. Basically, overshoots when using arisim sitl + arducopter is significantly larger then without. Why this can happen? and how to deal with that?

### Include context on what you are trying to achieve
I am using sitl drone and airsim as I need to do some image processing project through integrating Airsim, OpenCV and Dronekit. I was trying to first control the drone yaw myself using a joystick, which I managed to make 

I noticed that when I give input to rotate yaw, airsim copter rotates but does larger oscillations after it reaches the target then the case of using sitl alone without airsm **using exactly same code and same joystick**, just re-running the command of sitl in terminal is the difference

#### Context details
Windows 10 & Python 3.7
settings.json: https://gist.github.com/LightCannon/df453856d1c79887bed830707e539b5f

### Include details of what you already did to find answers
I tried searching here and in Ardupilot issues, but I cannot find anything related to this, specially that I see this behavior when using airsim sitl only

jonyMarino, 2022-02-24:
Hi and welcome @LightCannon!
I don't understand the test you did. Yes, the issue, not sure what is happening. Are you forced to use ardupilot? Can you see if the same happens if you use PX4?

LightCannon, 2022-02-24:
Basically, what I'm doing is rotating the drone around Yaw, using joystick & dronekit. I notice that the drone itself overshoots (don't settle quickly after I release the joystick input). However, this overshot is far too less then I use the sitl without airsim. 
Yes, I've to use Ardupilot.

samuelctabor, 2022-05-23:
ArduPilot needs some parameter changes to fly the default aircraft well, especially in yaw. These settings work a lot better for me.

ATC_RAT_YAW_P 0.2
ATC_RAT_YAW_I 0.2
ATC_RAT_YAW_D 0.09

I also made this change to fix the axis control mapping.

RCMAP_PITCH 4
RCMAP_YAW 2

I also found it flew better with reduced maximum rates and accelerations, but this is personal preference.

ATC_ACCEL_P_MAX  20000.000000
ATC_ACCEL_R_MAX  20000.000000
ATC_ACCEL_Y_MAX  27000.000000
ATC_RATE_P_MAX   100.000000
ATC_RATE_R_MAX   100.00000