microsoft/AirSim #4375: RPCError when calling simGetImages with several instances of ImageRequest

Villotadavid, 2022-02-25:
Hello!
I am trying to get two different images with the same instance of "simGetImages". For that I have written the following line:
```
responses = client.simGetImages(
[airsim.ImageRequest("CameraOF", 8,False,False), 
airsim.ImageRequest("CameraWide",0,True,False)], 
vehicle_name=VehicleName)
```
In the settings.json file I defined the following vehicle with its cameras as follows:

```
"Drone1": {
          "VehicleType": "SimpleFlight",
          "DefaultVehicleState": "Armed",
          "PawnPath": "",
          "EnableCollisionPassthrogh": false,
          "EnableCollisions": true,
          "AllowAPIAlways": true,
          "EnableTrace": false,
          "CameraWide": {
            "CaptureSettings": [
                {
                "ImageType": 0,
                "Width": 256,
                "Height": 144,
                "FOV_Degrees": 120
                }
            ],
            "X": 0, "Y": 0, "Z": 0
           },
           "CameraOF": {
            "CaptureSettings": [
                {
                "ImageType": 8,
                "Width": 256,
                "Height": 144,
                "FOV_Degrees": 45
                }
            ],
            "X": 0, "Y": 0, "Z": -0.5
           },
          "X": 0, "Y": 0, "Z": -2
        },
```

But whenever I run the code I always get the following error:
`msgpackrpc.error.RPCError: rpclib: function 'simGetImages' (called with 2 arg(s)) threw an exception. The exception contained this information: invalid map<K, T> key.`

I have tried several options having Airsim documentation as reference but I haven´t managed to run the code properly.

My current setup is:
Airsim: 1.5.0
Python: 3.9.7

rajat2004, 2022-02-26:
Cameras should be present inside the `Cameras` section, see https://microsoft.github.io/AirSim/settings/#vehicles-settings

Villotadavid, 2022-02-28:
Okay, so now I have downloaded some version 1.7.0 binaries and using the same code of line as in the main description I am getting the following error:
`msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 2, got: 3`
I do not understand why I am getting this error if I am using only two arguments, the first as a list and the second for the vehicle name.

Villotadavid, 2022-03-01:
Finally, after doing some research I found out that this is a version error.
When using v1.7.0. binaries, Airsim v1.6.0. should be used.
After updating to the last version, everything worked fine.

closed 2022-03-01 by Villotadavid

githjz, 2022-05-15:
Could you elaborate about how you solved the error please ?
You switched from which version of Airsim to which? And what do you mean by "When using v1.7.0. binaries, Airsim v1.6.0. should be used."?

I ran into the same error using Airsim v1.7.0, so I wonder if there is any other  solutions.  @Villotadavid

belongtothenight, 2022-06-04:
> 
I have encountered the same problem and successfully debugged it. As a result, I made two changes according to the author's comment above.
1. In cmd, type "pip install airsim==1.6.0". If successful, it will display "successfully uninstalled airsim-1.7.0" (assume you have 1.7.0 installed) and "successfully installed airsim-1.6.0."
2. Inside PyCharm (my current using IDE), go to file -> settings -> Project: "your project name" -> Python Interpreter -> + , search "airsim". On the bottom right corner check "Specify version", and type "1.6.0", and click the "Install Package" on the bottom left corner.

Ritv1k, 2024-12-16:
I did step one and now I am getting this error "msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 3, got: 2"