microsoft/AirSim #4381: Azure example doesn't work

rhalaly, 2022-02-26:
## Bug report 

- AirSim Version/#commit: 1.6.0
- UE/Unity version: 
- autopilot version:
- OS Version:

### What's the issue you encountered?

Running the Azure configuration causes an error when starting AirSim.

The only modify I did is to use the NV12s_V3 machines instead of NV12.

Got this message:
[image]

But NV12s_V3 has the Tesla M60 GPU, which support DX12 and Shader Model 6.4...

### Settings

Sefault

### How can the issue be reproduced?

1. Create Azure machine using the configuration in this repository
2. Choose NV12s_V3 as VM Size
3. Run AirSim

### Include full error message in text form
A D3D11-compatible GPU (Feature Level 11.0, Shader Model 5.0) is required to run the engine.

rhalaly, 2022-03-04:
After I reinstall NVIDIA drivers, I get:
```
DX11 feature level 10.0 is required to run the engine
```
But M60 support it and even further

rhalaly, 2022-03-07:
@jonyMarino can you assist with this issue, please?

jonyMarino, 2022-03-07:
@rhalaly I will take a look. Give me some time.

jonyMarino, 2022-03-08:
@rhalaly maybe some UE4 pre-requisites are missing on the machine.
Can you check/reinstall DirectX runtime on the machine and try again?

rhalaly, 2022-03-08:
@jonyMarino I took the https://github.com/microsoft/AirSim/blob/master/azure/azure-env-creation/configure-vm.ps1 to configure the VM. I tried to reinstall DirectX but same error appears

zimmy87, 2022-03-18:
Hey @rhalaly, we're trying to repro this issue. One thing we found so far is that in order to start the deployment, we had to change this setting in the template: 

```
    "resources": [
        {
            "type": "Microsoft.Compute/virtualMachines",
            "apiVersion": "2019-07-01",
            "name": "[parameters('VmName')]",
            ...
            "properties": {
                ...
                "storageProfile": {
                    "imageReference": {
                        "publisher": "MicrosoftWindowsDesktop",
                        "offer": "Windows-10",
                        "sku": "rs5-pro",
                        "version": "latest"
                    },
                    ...
                },
               ...
            }
```

In this setting, we had to change the value of "resources" -> "properties" -> "storageProfile" -> "imageReference" -> "sku" from "rs5-pro" to "win10-21h2-pro".

While the deployment is failing for us, we were able to get the Blocks binary to run after manually running the configure-vm.ps1 script on the VM. Unfortunately, we only currently have access to NV12 VM size in the subscription and region we're using. Would you mind telling us which region you are using to get access to the NV12s_V3 VM size?

rhalaly, 2022-03-18:
Thanks for the update @zimmy87! I'm using East US. I changed the OS to Windows Server 2019.

But with NV4as_V4 (AMD GPU) I was able to use Blocks. I suspect that this issue is related to GPU drivers/OS compatibility

zimmy87, 2022-03-18:
Hey @rhalaly, that's good to hear. Are you still interested in using the NV12s_V3 VM size or is NV4as_V4 sufficient for your purposes going forward?

rhalaly, 2022-03-18:
Yes, I prefer NV12s_V3 there is more flexibility with Nvidia GPUs

zimmy87, 2022-04-07:
Hey @rhalaly, did you by chance manually install Nvidia drivers on your Azure machine? If so, there's a possibility you may not be using the GRID drivers that have the license for running in WDDM graphical rendering mode instead of TCC CUDA compute mode. Can you check whether your drivers are in WDDM mode?

UnfixedMold, 2024-05-27:
any updates on this?