microsoft/AirSim #4383: Image Tile Error

ayush055, 2022-03-02:
## Question
### What's your question?

When running the dqn_car.py or dqn_drone.py, I am getting the following error:

**ValueError: tile cannot extend outside image**

This error occurs when this line is run: **image = Image.fromarray(img2d)**

In addition, this error first occurs after many timesteps pass during training.

jonyMarino, 2022-03-09:
@[ayush055](https://github.com/ayush055) which AirSim version and wich environment? settings.json?

ayush055, 2022-03-12:
Airsim version: 1.7.0
OS: Windows 11

Settings.json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2
}

jonyMarino, 2022-03-14:
And the unreal environment you are using?

NNaroa, 2022-03-29:
Same problem over here. The error occurs sometimes after 5 minutes of training, sometimes after more than 12 hours of training (without changing anything in the scripts or the environment). 

I've tried several configurations:
- Windows 10, Airsim 1.6.0, UE 4.27.2, Blocks environment.
- Windows 10, Airsim 1.6.0, UE 4.27.2, LandscapeMountains environment.
- Windows 10, Airsim 1.7.0, UE 4.27.2, Blocks environment.
- Windows 10, Airsim 1.7.0, UE 4.27.2, LandscapeMountains environment.

For all of them, the settings.json was the following:

`{
  "SeeDocsAt": "https://microsoft.github.io/AirSim/settings/",
  "SettingsVersion": 1.2,
  "ClockSpeed": 100,

  "SimMode": "Multirotor",
   "Vehicles": {
      "Drone1": {
          "VehicleType": "simpleflight",
          "AutoCreate": true,
          "Sensors": {
            "LidarSensor1": {
                "SensorType": 6,
                "Enabled" : true,
                "NumberOfChannels": 32,
                "Range": 20,
                "RotationsPerSecond": 10,
                "PointsPerSecond": 100000,
                "X": 0, "Y": 0, "Z": -1,
                "Roll": 0, "Pitch": 0, "Yaw" : 0,
                "VerticalFOVUpper": 10,
                "VerticalFOVLower": -10,
                "HorizontalFOVStart": -30,
                "HorizontalFOVEnd": 30,
                "DrawDebugPoints": true,
                "DataFrame": "SensorLocalFrame"
            }
          }
      }
  }
}`

jackvice, 2022-11-16:
Did anyone find a solution to this?

praingeard, 2023-03-23:
Hello, has anyone found a solution to this problem?