microsoft/AirSim #4388: start airism cgywin px4 sitl, In 10-30 minutes the plane will crash out of control

d137578736, 2022-03-04:
## Bug report 

- AirSim Version/#commit: 1.6.0
- UE/Unity version:UE4.6.2
- autopilot version:1.3.0
- OS Version:win10
- QGC version: 4.1.13

### What's the issue you encountered?

When multi-rotor is take off,Don't do anything,In 10-30 minutes the plane will crash out of control
### Settings

{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockType": "SteppableClock",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "LockStep": true,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPortLocal": 14540,
            "ControlPortRemote": 14580,
            "Sensors":{
                "Barometer":{
                    "SensorType": 1,
                    "Enabled": true,
                    "PressureFactorSigma": 0.0001825
                }
            },
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165
            }
        }
    }
}
### How can the issue be reproduced?

1.Start cgywin PX4
2.Start airism,Airsim will link px4 success
3.Start qgc ,use usb RC controll,adjusting RC controll
4.User RC controll take off PX4 is success
5.Wait,In 10-30 minutes the plane will crash out of control

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-03-04:
Hi @d137578736 and welcome!  The recommended platform for PX4 is currently WSL2

HarrySoteriou, 2022-03-05:
Follow this tutorial to setup PX4 with wsl2 amd AirSim https://youtu.be/DiqgsWIOoW4

d137578736, 2022-03-07:
> Hi @d137578736 and welcome! The recommended platform for PX4 is currently WSL2

think you very much!  i will test airism in wsl2 environment!

d137578736, 2022-03-07:
> Follow this tutorial to setup PX4 with wsl2 amd AirSim https://youtu.be/DiqgsWIOoW4
think you very much!