microsoft/AirSim #4389: The interval between adjacent IMU measurements varies.

lnexenl, 2022-03-04:
## Bug report 

- AirSim Version/#commit: v1.6.0
- UE/Unity version: UE v4.27.2
- autopilot version:
- OS Version: Manjaro

### What's the issue you encountered?
I record a rosbag with 300hz imu data. However I found that the interval between adjacent IMU measurements varies, that's say the variance of the interval is big.

I have printed some timestamp of the IMU measurements:
```
1643944932643415681
1643944932643443757
1643944932643451510
1643944932643510595
1643944932643521560
1643944932643526728
1643944932643530919
1643944932643622341
1643944932643670881
1643944932643685198
1643944932687163209
1643944932687199457
1643944932687204206
1643944932687209444
1643944932687220828
1643944932687260987
1643944932687271393
1643944932687293463
1643944932687301914
1643944932687354993
1643944932687358345
1643944932687385304
1643944932687390961
1643944932731163772
1643944932731179276
1643944932731210076
1643944932731218527
1643944932731231099
1643944932731242413
1643944932731245765
1643944932731248559
1643944932731289695
1643944932731299194
1643944932731302057
1643944932731306806
1643944932731333556
1643944932731343473
1643944932775119147
1643944932775169782
1643944932775175439
```
We can see that there will be a jump every several measurements. I think varying intervals will make 300hz measurements perform like equally-sampled 50hz or lower frequency measurements.

zimmy87, 2022-03-18:
I tried to repro this locally by placing a `msr::airlib::Utils::log()` call inside `ImuSimple::updateOutput()`, however my console output showed a consistent time delta of ~0.003s between IMU measurements (this varied slightly with different ClockTypes: ScalableClock showed slight, negligible variance each iteration, whereas SteppableClock showed 0.003 consistently), which is the same as the default physics update rate (defined as `physics_loop_period_` & set [here](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldBase.h#L73)). This leads me to believe you might be experiencing 1 of 2 possible different situations:

- You mentioned you're using a rosbag to record your IMU measurements: there's a possibility that the update rate of the project code calling the AirSim API to read the IMU measurements may be lower than the internal update rate of the IMU. Are you able to provide the relevant project code doing the reading? Are you currently using the ROS wrapper to facilitate this? If so, is it ROS 1 or 2?
- You may have performance issues with AirSim: inside AirSim, you can display your FPS by opening the console with the \` key, then entering `stat FPS`. I would be curious to know if you're seeing occasional dips in FPS. If you are, you may be able to work around this by changing your ClockType (I suspect SteppableClock may be the better choice here).

zimmy87, 2022-03-21:
I dug a little deeper in this issue and tried to repro this using a rosbag recording the `/airsim_node/SimpleFlight/imu/imu` rostopic published by the ros wrapper with Windows 11 + Ubuntu  20.04 in WSL2. I found each message on this rostopic had a timestamp with a time delta of about ~0.01s after the previous message. This is expected due to the publishing frequency set in the airsim_node launch file [here](https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch#L11). So unfortunately, I am currently unable to reproduce the behavior you described with your imu data. Is using WSL2 suitable for your scenario?

lnexenl, 2022-03-22:
I am currently using ROS1(noetic) wrapper to record the bag. Currently I changed the IMU record freq to 200Hz, but the same situation still occures. My currently airsim_node.launch:
```xml
<launch>
        <arg name="output" default="log"/>
        <arg name="publish_clock" default="true"/>
        <arg name="is_vulkan" default="false"/>
        <arg name="host" default="localhost" />

        <node name="airsim_node" pkg="airsim_ros_pkgs" type="airsim_node" output="$(arg output)">
                <param name="is_vulkan" type="bool" value="false" /> 
                <!-- ROS timer rates. Note that timer callback will be processed at maximum possible rate, upperbounded by the following ROS params -->
                <param name="update_airsim_img_response_every_n_sec" type="double" value="1000" /> 
                <param name="update_airsim_control_every_n_sec" type="double" value="0.005" />
                <param name="update_lidar_every_n_sec" type="double" value="1000" />
                <param name="publish_clock" type="bool" value="$(arg publish_clock)" />
                <param name="host_ip" type="string" value="$(arg host)" />
        </node>

        <!-- Static transforms -->
        <include file="$(find airsim_ros_pkgs)/launch/static_transforms.launch"/>
</launch>
```
and my current settings.json:
```json
{
    "SettingsVersion": 1.2,

    "SimMode": "Car",
    "ClockType": "SteppableClock",
    "Vehicles": {
        "car":{
            "VehicleType": "PhysXCar",
            "Sensors": {
                "Barometer": {
                    "SensorType": 1,
                    "Enabled" : true,
                    "PressureFactorSigma": 0.001825,
                    "PressureFactorTau": 3600,
                    "UncorrelatedNoiseSigma": 2.7,
                    "UpdateLatency": 0,
                    "UpdateFrequency": 50,
                    "StartupDelay": 0
                },
                "Imu": {
                    "SensorType": 2,
                    "Enabled" : true,
                    "AngularRandomWalk": 0.3,
                    "GyroBiasStabilityTau": 500,
                    "GyroBiasStability": 4.6,
                    "VelocityRandomWalk": 0.24,
                    "AccelBiasStabilityTau": 800,
                    "AccelBiasStability": 36
                },
                "Gps": {
                    "SensorType": 3,
                    "Enabled" : true,
                    "EphTimeConstant": 0.9,
                    "EpvTimeConstant": 0.9,
                    "EphInitial": 25,
                    "EpvInitial": 25,
                    "EphFinal": 0.1,
                    "EpvFinal": 0.1,
                    "EphMin3d": 3,
                    "EphMin2d": 4,
                    "UpdateLatency": 0.2,
                    "UpdateFrequency": 50,
                    "StartupDelay": 1
                },
                "Magnetometer": {
                    "SensorType": 4,
                    "Enabled" : true,
                    "NoiseSigma": 0.005,
                    "ScaleFactor": 1,
                    "NoiseBias": 0,
                    "UpdateLatency": 0,
                    "UpdateFrequency": 50,
                    "StartupDelay": 0
                }
            }
        }
    }
}

```
But using previous settings I still find that intervals vary. I use `rosbag hz` to show the freq of IMU topic, and the output is:
```
average rate: 199.931
        min: 0.001s max: 0.007s std dev: 0.00069s window: 200
average rate: 200.023
        min: 0.001s max: 0.007s std dev: 0.00063s window: 400
average rate: 199.987
        min: 0.001s max: 0.007s std dev: 0.00062s window: 600
average rate: 200.008
        min: 0.001s max: 0.007s std dev: 0.00061s window: 800
average rate: 200.013
        min: 0.001s max: 0.007s std dev: 0.00060s window: 1000
average rate: 199.979
        min: 0.001s max: 0.007s std dev: 0.00059s window: 1201
average rate: 200.011
        min: 0.001s max: 0.007s std dev: 0.00059s window: 1401
average rate: 199.993
        min: 0.001s max: 0.007s std dev: 0.00059s window: 1601
average rate: 199.980
        min: 0.001s max: 0.007s std dev: 0.00059s window: 1801
average rate: 199.988
        min: 0.001s max: 0.007s std dev: 0.00059s window: 2002
```
The std dev is still a bit big.

And as for performance problem, I use `stat FPS` to show the fps and it's floating between 70 to 100fps, so I think there is no performance problem with my PC?

And I will try WSL2 later to see whether it can solve the problem, thank you!

lnexenl, 2022-03-28:
Okay I will do the setting and then take a test to see whether RPC have some latency and update the result here. Thank you! @zimmy87

lnexenl, 2022-09-13:
Ah I think I have found where the problem is now. I dig into the source code, and found that: Different from multirotor, car uses **PhysX** as physics engine, instead of physics engines implemented by AirSim. Sensor reading changes when PhysX "Tick"s. Unreal's PhysX engine ticks once every rendered frame as default. So the update frequency is same as fps as default.

I have recorded some 200Hz IMU data, and I found that the reading changes about every 0.015s, that's quite near my monitor's fresh rate(60Hz)'s reciprocal, which can verify this statement.

I find some links that might be useful for solving this problem:
https://forums.unrealengine.com/t/using-a-fixed-physics-timestep-in-unreal-engine-free-the-physics-approach/67537/7

https://docs.unrealengine.com/4.27/en-US/InteractiveExperiences/Physics/Substepping/ 

I will try these methods later and update my progress here.