microsoft/AirSim #4392: Adding Airsim Plugin to LandscapeMountains Unreal Environment

tristanthomas3, 2022-03-05:
Hello all, so I am using this video https://www.youtube.com/watch?v=1oY8Qu5maQQ as guidance to add the AirSim plugin to landscape mountains. I can get all the way through all the steps, however whenever I try to open MyProject from the Unreal editor I get this error

"The following modules are missing or built with a different engine version: LandscapeMountains Would you like to rebuild them now?"

I opened the status file and see the only error looks like "LogInit: Warning: Incompatible or missing module: LandscapeMountains"

Anyone know what I need to do to solve this? When I click rebuild I get another error.

HarrySoteriou, 2022-03-05:
Hey just out of curiosity did you build Unreal from source or did you download the binaries only? I tried using the binaries (smaller version) instead of building from source on one computer and faced similar issues. I will check if it works this time and get back to you. Also if you do decide to build from source go to Unreal Engine forums to see how to unselect UE build for unecessary OS (during the Setup command you can exclude build targets such as Android, Mac, etc...) to significantly reduce build size and safe time (build size should be around 100Gb if you build for all targets)

tristanthomas3, 2022-03-05:
I downloaded Unreal directly from the epic games launcher how it was described in the documentation on the AirSim GitHub page, I am honestly not sure which one that is. But yes please do let me know.

HarrySoteriou, 2022-03-05:
Installing from source worked but took so long as I hadnt unsellected any targets when I ran the Setup last night (total size was around 116GB if I remember correctly, 84GB for plugins) which is obsurd!

Installing Unreal engine from the Epic Games Launcher does not work with AirSim, as I tried to do that repeatedly.

The safest way is to:

1) Install Epic Games Launcher
2) git clone release UE repository
3) Build all of the targets (I am not 100% sure that is okay to exclude them during build but if it is then that would save alot of time)
4) After building as per their instructions on their page rename the folder to something like UE_4.27 and place it under C:/Program Files/Epic Games/
5) Then open Epic Game Launcher > go to UE engine version> click on the drop down menu next to it> Options> Unselect all other OS (android, Mac, Linux,...) this will remove 30-40Gb
6) Also underneath your engine version there is an option " installed plugins" which you can click and then scroll down to remove unwanted plugins to save space ( I do not know which plugins are essential)

It's just a shame that you have to figure these things out by trial and error wasting so many hours trying to get it to work.

tristanthomas3, 2022-03-06:
I don't think that is true. I know others who have installed Unreal from Epic Games Launcher and were able to successfully add in the AirSim plugin.

The error I get is in the FAQ, but the only solution given is not the issue that I am having.

HarrySoteriou, 2022-03-06:
Hmm that's interesting. I could be wrong but everytime I tried building AirSim without using the source version I couldn't. Please let me know if you manage to build it without downloading the source version

jonyMarino, 2022-03-09:
Hi @tristanthomas3! Can you follow [this instructions](https://microsoft.github.io/AirSim/unreal_custenv/)?

HarrySoteriou, 2022-03-10:
> Hi @tristanthomas3! Can you follow [this instructions](https://microsoft.github.io/AirSim/unreal_custenv/)?

Dear @jonyMarino when was the last time you got it to work though these instructions without using the source code build ?

I have been trying again and again (for weeks!) using the one downloaded from the Epic Launcher (20.1GB) and I can never get it to work. 

Repeating the same steps with the source build leads to a successful build of AirSim, but the size is around 115GB before removing build from unnecessary OS's and even after that due to all the Plugins being built I am left with an UE4 of 86GB.

This is a very serious issue. Please respond after replicating the steps.

tristanthomas3, 2022-03-15:
@jonyMarino I agree, I think the issue described is different. To be more clear, I did follow those instructions, or at least attempt to do so prior to looking at and utilizing the video. Ultimately, I got the same error using those instructions as I did using the video. Specifically, when pressing F5 to run as described in step 10 of the instructions, I get the same build error: "The following modules are missing or built with a different engine version: LandscapeMountains Would you like to rebuild them now?" When I click yes to rebuild it tells me I will have to rebuild from source manually. This issue is happening for me regardless of whether I try and build using the video or the written instructions that you attached.

jonyMarino, 2022-07-25:
I am adding LandscapeMountains from [this repository](https://github.com/jonyMarino/LandscapeMountains) into the Environments folder and building with build.cmd works just fine.

PR #4400 (2022-03-10): Update unreal_custenv.md

Fixes: #4392    

## About

Landscape Mountains assets don't support UE 4.25. The video's author was copying assets to a new project, that approach doesn't work anymore. The instructions on the documentation should be followed instead (using the project itself and not copying and pasting assets)

## How Has This Been Tested?

## Screenshots (if appropriate):
PR #4401 (2022-03-11): add note about migrating the Landscape Mountains project

Fixes: #4392    

## About
The Landscape Mountains project referenced in docs/unreal_custenv.md is only supported up to UE 4.24. This PR adds a note about migrating this project before adding the AirSim plugin.

## How Has This Been Tested?

## Screenshots (if appropriate):