microsoft/AirSim #4398: Not working as intended simSwapTextures, simSetObjectMaterial, simSetObjectMaterialFromTexture

HSC-1, 2022-03-09:
## Question
### What's your question?

First
I Failed to use simSwapTextures 
Following the [example](https://microsoft.github.io/AirSim/retexturing/), I returned empty list

```
import airsim 

c = airsim.VehicleClient()
print(c.simSwapTextures("test",0))
```
[image]
[image]
So I'm looking for replacement
find #3992 
I try this API but not working as intended 

```
print(c.simSetObjectMaterialFromTexture("texture", "sample_texture.jpg"))
print(c.simSetObjectMaterial("texture","sample_texture"))
```
This output is True, False  
[image]

I obviously care about directories 
sample_texture.jpg in my diretory is with python file
Jist in case, I put it all diretory else

HSC-1, 2022-03-31:
Add Visual Studio log message 
```
LogStreaming: Warning: Failed to read file 'sample_texture.jpg' error.
LogImageUtils: Warning: Error creating texture. sample_texture.jpg could not be found
```
I found Visual Studio log message. 
What settings should UE4 set so that the above errors are not occur?
I'm sure I put the image file in all the directories.

HSC-1, 2022-04-01:
Sorry path problem and meterial string proble 
But if it's not blueprint, it doesn't work as intended.
The chair is blueprint,Just the table is static mesh actor.
[image]

closed 2022-04-07 by HSC-1

deoliveirajoshua, 2022-05-25:
I'm getting a similar error when running the change_texture_example.py script in UE 4.27 version. The jpg is in the working directory of my script. How did you find a workaround?

HSC-1, 2022-05-25:
Hello, can you try by modifying the full directory of jpg?
ex) "C:./user/Airsim/image.jpg"

tmddn833, 2023-07-05:
Does 

>```
>print(c.simSetObjectMaterialFromTexture("texture", "sample_texture.jpg"))
>print(c.simSetObjectMaterial("texture","sample_texture"))
>```
>This output is True, False

Have you ever seen simSetObjectMaterial function work well?
Changing texture to simSetObjectMaterialFromTexture at runtime using jpg of absolute path was successful.
However, despite various attempts so far, no reflection has been made using the materials in the asset.
Below are the things I tried.

1) c.simSetObjectMaterial("texture","{absolute path to jpg}") -> False
2) c.simSetObjectMaterial("texture","{path copied from unreal editor by "copy reference"}") -> False
2) c.simSetObjectMaterial("texture","{path copied from unreal editor by "copy file path"}") -> False
[image]

If there is anyone who know how to change the material with asset in the unreal editor, please leave comment.
Thank you