microsoft/AirSim #4413: Move one meter in x direction

krishnarajk, 2022-03-15:
I am trying to move the drone in unreal engine using MoveToPositionAsync function. I want to move the drone 1 meter in the x-direction. What value needs to be given as a parameter in the MoveToPositionAsync function in order to get the expected result. When I give x value as 1, it is moving ~123 uu.
Thanks

jonyMarino, 2022-03-22:
Hi @krishnarajk! The moveToPositionAsync uses the world frame, not the vehicle one. So, where is your drone located? 
Also, you have to consider that there is a lookahead error. Take a look at #4293

krishnarajk, 2022-03-22:
@jonyMarino Thanks for the reply. Here is the location of drone in different states.
Before takeoff
[image]

After takeoff
[image]

After moving in x axis, x value given as 1.
[image]

krishnarajk, 2022-03-22:
So basically it should move exactly 1 meter? Is it because of the lookahead error?

jonyMarino, 2022-03-22:
Yes, it is. You can try to decrease it, but it can lead to unwanted behavior in other use cases.

krishnarajk, 2022-03-23:
How to reduce it?
I tried reducing the value in the "virtual float getDistanceAccuracy()" (from 0.5 to 0.1) function but didn't work for me as expected. 
[image]

That x value increases to 1.73

jonyMarino, 2022-04-06:
@krishnarajk, sorry for the delay. It doesn't seem obvious to me to fix the problem by modifying our flight controller. Other alternatives are PX4 or ardupilot. Also, our ROS wrapper has its own PD controller to set the position, and you can adjust its parameters. Do any of these alternatives work for you?