microsoft/AirSim #4418: LiDAR accuracy incorrect

thomasTUK, 2022-03-17:
## Bug report 

- AirSim Version/#commit: N/A latest probably
- UE/Unity version: UE 4.26
- autopilot version: simple_flight
- OS Version: Windows 10

### What's the issue you encountered?
I have run a point cloud simulation, because I want to calculate the point spaces beforehand the simulation.

[image]
[image]

During the experiment, I have notived that the point cloud during a single shot produces twice as many points as it should be. As you can see in the screenshots. 

[image]
[image]

Point Cloud is here:
[points_14_25_48_0_0_-2.txt](https://github.com/microsoft/AirSim/files/8286370/points_14_25_48_0_0_-2.txt)

### Settings
            "LidarSensor1": {
                "SensorType": 6,
                "Enabled" : true,
                "NumberOfChannels": 16,
                "Range": 100,
                "PointsPerSecond": 288000,
                "RotationsPerSecond": 20,
                "HorizontalFOVStart": 0,
                "HorizontalFOVEnd": 0,
                "VerticalFOVUpper": 15,
                "VerticalFOVLower": -15,
                "X": 0,
                "Y": 0,
                "Z": -1,
                "Roll": 0,
                "Pitch": 0,
                "Yaw" : 0,
                "DataFrame": "SensorLocalFrame",
                "DrawDebugPoints": true

### How can the issue be reproduced?

1.  Run a test with drone_lidar.py and write to disk

zimmy87, 2022-03-25:
Can you elaborate on how you're selecting points in your third screenshot? It looks like you're using CloudCompare, but it's unclear to me how you're selecting a subset of points that corresponds to those produced by a "single shot". I did some testing locally with the settings you provided and with the default update frequency of 10, one single iteration of the lidar obtaining a point cloud should do 29377 line traces (see [here](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp#L55) for where this is calculated), so you should see a lot more than just 4 points for a single shot. Also, the timestamp at which a point is detected isn't stored in the point cloud, so I don't know of a way for selecting the subset for a single shot other than manually counting a certain number.

thomasTUK, 2022-03-25:
HI @zimmy87 , Thanks for the message! The third picture is just a zoom in the second picture. 
Can you provide me some code of how to integrate the time stamps?

zimmy87, 2022-04-05:
The timestamp is included in the lidar output [here](https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/lidar/LidarSimple.hpp#L101). You can read this value by accessing the `LidarData.time_stamp` field returned by a call to `getLidarData` in the python client. For example, in a python script, you can print the timestamp by executing the following (not tested locally):

```
import airsim

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)

lidarData = client.getLidarData()
print(lidarData.time_stamp)
```

thomasTUK, 2022-04-06:
I'll have a look @zimmy87! Are you able to answer my question, posted here? It is regarding the timestamp but also the Lidar parameters.
https://github.com/microsoft/AirSim/discussions/4406