microsoft/AirSim #4422: Airsim stuck after connecting to jupyter/E2EDeepLearning #URGENT

danialvi, 2022-03-18:
## Question
### What's your question?
So I have been working on E2EDeepLearning for Airsim for over a month now, and struggling to go further. I have tried looking up this issue without any luck. After training the model with jupyter, and connecting with Airsim, jupyter gets stuck and doesn't respond. Neither do airsim. 

### Include context on what you are trying to achieve
This is the guidelines I'm following: https://github.com/microsoft/AutonomousDrivingCookbook/tree/master/AirSimE2EDeepLearning

#### Context details
 
[settings-json.zip](https://github.com/microsoft/AirSim/files/8307905/settings-json.zip)

Tried looking up in the github repo issue directory. Didn't find any solution that worked, also tried to contact with others to solve the issue. 

[1] 
This is the testing notebook that fails after connecting: 
from keras.models import load_model
import sys
import numpy as np
import glob
import os

if ('../../PythonClient/' not in sys.path):
    sys.path.insert(0, '../../PythonClient/')
from AirSimClient import *

# << Set this to the path of the model >>
# If None, then the model with the lowest validation loss from training will be used
MODEL_PATH = None

if (MODEL_PATH == None):
    models = glob.glob('model/models/*.h5') 
    best_model = max(models, key=os.path.getctime)
    MODEL_PATH = best_model
    
print('Using model {0} for testing.'.format(MODEL_PATH))

[2] 
model = load_model(MODEL_PATH)

client = CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = CarControls()
print('Connection established!')

[*]
car_controls.steering = 0
car_controls.throttle = 0
car_controls.brake = 0

image_buf = np.zeros((1, 59, 255, 3))
state_buf = np.zeros((1,4))

Need help asap

kutilanak, 2022-04-09:
did u run the .py or .ipynb ?

try to run the whole code not 1 by 1

danialvi, 2022-04-10:
> did u run the .py or .ipynb ?
> 
> try to run the whole code not 1 by 1

I did try one by one in jupyter. Should i try by running all at once in jupyter or run the file in miniconda?

kutilanak, 2022-04-10:
> > did u run the .py or .ipynb ?
> > try to run the whole code not 1 by 1
> 
> I did try one by one in jupyter. Should i try by running all at once in jupyter or run the file in miniconda?

I think whatever app you're using didn't matter as long as you're running it all at once

[image]

danialvi, 2022-04-11:
> > > did u run the .py or .ipynb ?
> > > try to run the whole code not 1 by 1
> > 
> > 
> > I did try one by one in jupyter. Should i try by running all at once in jupyter or run the file in miniconda?
> 
> I think whatever app you're using didn't matter as long as you're running it all at once
> 
> [image]

THANK YOU!!! IT worked !