microsoft/AirSim #4441: Cannot run blocks environment

kutilanak, 2022-03-28:
## Question
### What's your question?

I've followed the tutorial here [https://microsoft.github.io/AirSim/build_windows/](url) and here [https://microsoft.github.io/AirSim/unreal_blocks/](url) but it seems I still can't run Blocks Environment successfully.

### Include context on what you are trying to achieve

I'm trying to run Blocks Environment.

#### Context details

Airsim 1.6.0
VS2019
UE 4.24

error msg:
> 1>------ Build started: Project: Blocks, Configuration: DebugGame_Editor x64 ------
> 1>Creating makefile for BlocksEditor (no existing makefile)
> 1>G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs : warning : Referenced directory 'G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\eigen3' does not exist.
> 1>G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs : warning : Referenced directory 'G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\include' does not exist.
> 1>G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs : warning : Referenced directory 'G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\rpclib\include' does not exist.
> 1>UnrealBuildTool : warning : Library 'Shell32.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
> 1>UnrealBuildTool : warning : Library 'dinput8.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
> 1>UnrealBuildTool : warning : Library 'dxguid.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
> 1>UnrealBuildTool : warning : Library 'G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
> 1>UnrealBuildTool : warning : Library 'G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\lib\x64\Release\AirLib.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
> 1>UnrealBuildTool : warning : Library 'G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\rpclib\lib\x64\Release\rpc.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
> 1>Parsing headers for BlocksEditor
> 1>  Running UnrealHeaderTool "G:\new\AirSim\Unreal\Environments\Blocks\Blocks.uproject" "G:\new\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\BlocksEditor\DebugGame\BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
> 1>Reflection code generated for BlocksEditor in 6.8004173 seconds
> 1>Building BlocksEditor...
> 1>Using Visual Studio 2019 14.29.30141 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133) and Windows 10.0.18362.0 SDK (C:\Program Files (x86)\Windows Kits\10).
> 1>Building 11 actions with 16 processes...
> 1>  [1/11] Default.rc2
> 1>  [2/11] Default.rc2
> 1>  [3/11] SharedPCH.Engine.NonOptimized.ShadowErrors.cpp
> 1>  [4/11] Blocks.cpp
> 1>  [5/11] UE4Editor-Blocks-Win64-DebugGame.lib
> 1>     Creating library G:\new\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\Blocks\UE4Editor-Blocks-Win64-DebugGame.lib and object G:\new\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\Blocks\UE4Editor-Blocks-Win64-DebugGame.exp
> 1>  [6/11] Module.AirSim.cpp
> 1>G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\VectorMath.hpp(14): fatal error C1083: Cannot open include file: 'Eigen/Dense': No such file or directory
> 1>  [7/11] Module.AirSim.gen.cpp
> 1>G:\new\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\VectorMath.hpp(14): fatal error C1083: Cannot open include file: 'Eigen/Dense': No such file or directory
> 1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(46,5): error MSB3075: The command "G:\Unreal\UE_4.24\Engine\Build\BatchFiles\Build.bat BlocksEditor Win64 DebugGame -Project="G:\new\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
> 1>Done building project "Blocks.vcxproj" -- FAILED.
> ========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

### Include details of what you already did to find answers
N/A

Thank you!

jonyMarino, 2022-03-28:
It seemes you are mising some dependencies. Didn't you get any error on the previous steps?

kutilanak, 2022-03-29:
> It seemes you are mising some dependencies. Didn't you get any error on the previous steps?

I tried to install UE 2.5, build AirSim again, and follow this step again https://microsoft.github.io/AirSim/unreal_blocks/ and it works for me. 

Thank you for your response! 

I'm closing this thread

closed 2022-03-29 by kutilanak