microsoft/AirSim #4442: AirSim (+ ROS wrapper) not publishing clock when /use_sim_time is set to true?

zdjeffri, 2022-03-28:
## Bug report 

- AirSim Version/#commit: commit 8057725712c0cd46979135396381784075ffc0f3
- UE/Unity version: 4.27.2-18319896
- autopilot version: N/A
- OS Version: Windows 11 21H2, Ubuntu 20.04.4, ROS Noetic, (Using WSL2)

### What's the issue you encountered?
Using AirSim to simulate lidar data for an autonomous robot. Have the clock speed for AirSim set at 0.5 (via settings.json). Can confirm this works properly because the clock topic is running at half-speed when the "use_sim_time" rosparam is false. 
However, if "use_sim_time" is set to true, the clock is no longer published and the ROS Wrapper ceases to publish anything. In order to properly integrate the AirSim clock with other ROS nodes, "use_sim_time" needs to be set to true so that the ROS Time API returns what is being published by AirSim instead of the system clock. 
 
[image]

### Settings and other relevant files
[AirSim Settings - settings.json](https://gist.github.com/zdjeffri/e6630e07ed8e8615cc5233d4f152f8a8)
[roslaunch file - airsim_node.launch](https://gist.github.com/zdjeffri/c1056923100c9b7ebd69c345a71cc565)

### How can the issue be reproduced?
1. Apply the settings.json
      - the only pertinent settings are the host and clock speed. the vehicle doesn't matter. 
 2. Adjust the launch file
      - publish_clock is set to true
           - because it is a private nodehandle publisher, I also remap is from /airsim_node/clock to /clock. This isn't pertinent to the issue though.
      - use_sim_time is set to true
           - required so that other ros nodes run on same clock as AirSim
 3. Launch the file: `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP`
 4. press play in AirSim in separate unreal window, wait for the AirSim ros node to connect
 5. rostopic echo any of the AirSim topics - nothing is being published

zdjeffri, 2022-03-28:
I'll add, that in ROS terms at least, the airsim_node is not acting as a clock server when "use_sim_time" is true, but when "use_sim_time" is false, it is not acting as a clock server. It should work in both cases, or at least the opposite case. Hopefully, this clarifies the issue.

yhabib29, 2022-03-31:
Hello,
I also had an issue using ROS wrapper regarding simulated clock. I was recording all data into a rosbag.
To be able to record simulated data and clock while using a custom _ClockSpeed_ I did the following:
1. In **`airsim_node.launch:7 `** I added:
```
<param name="/use_sim_time" value="true" if="$(arg publish_clock)">
```
2. Thus, I launch airsim_node using following command:
```
roslaunch airsim_ros_pkgs airsim_node.launch publish_clock:=true
```
3. Renamed clock topic from `/airsim_node/clock` to `/clock`. To do this, I replaced _NodeHandle_ `nh_private_` to `nh_` in **`airsim_ros_wrapper.cpp:319`** :
```
clock_pub_ = nh_.advertise<rosgraph_msgs::Clock>("/clock", 1);
```

This way, my rosbag is recorded using published clock.
When I play the rosbag using `--clock` option everything works well ...

zdjeffri, 2022-03-31:
@yhabib29 What version of AirSim are you using? Are you using a car or a drone? What version of ROS?
This solution did not work. AirSim, whether using public or private nodehandle, only publishes clock is use_sim_time is false - which is my problem. If use_sim_time is true, clock is not published.

dylanyd, 2023-08-18:
I recently meet the same problem and solve it by changing the timer of drone_state_timer_cb from ros::Timer to ros::WallTimer.

andrewjong, 2024-01-17:
The issue seems to be that AirSim's ROS wrapper publishes `/airsim_node/clock` based on `ros::Timer`, which when `use_sim_time=true` only runs if `/clock` is advancing. So it's a chicken-egg problem.

```cpp
// airsim_ros_wrapper.h
    ros::Timer airsim_control_update_timer_;

// airsim_ros_wrapper.cpp
    airsim_control_update_timer_ = nh_private_.createTimer(
            ros::Duration(update_airsim_control_every_n_sec), 
            &AirsimROSWrapper::drone_state_timer_cb, this);

void AirsimROSWrapper::drone_state_timer_cb(const ros::WallTimerEvent& event)
{
    // other stuff
        // publish the simulation clock
        if (publish_clock_) {
            clock_pub_.publish(ros_clock_);
        }
   // other stuff
}

```

@dylanyd's `ros::WallTimer` solution is a valid workaround that makes `airsim_control_update_timer_` advance from the system clock instead of ROS' `/clock` topic. However a fully correct solution would be to somehow sync to AirSim/Unreal Engine's internal clock. That way setting AirSim's clockspeed will also synchronize here. Not sure how to do that yet. Will update when that's figured out.