microsoft/AirSim #4447: ROS wrapper GPU memory leak

yhabib29, 2022-03-30:
## Bug report 

- AirSim Version/#commit: v1.7.0-linux
- UE/Unity version: 4.25.4
- autopilot version: PX4 
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

I use AirSim with Unreal Engine to simulate a drone with a stereo camera. I am using DJI S900 model and a combination of few settings I have found on the AirSim doc. The simulation launches, I can control using the PX4 autopilot and the Python API.
However, when I want to retrieve camera images using AirSim ROS wrapper, I observe a memory leak, the GPU memory increases until it completely saturates.
The memory Ieak even occurs with a single camera ("front_left_custom"), it just take more time to saturate. 
Before launching AirSim ROS wrapper:
[image]
After launching AirSim ROS wrapper:
[image]

### Settings

[settings.json](https://gist.github.com/yhabib29/8639635bc7656f9da669517df5ba1c66)

### How can the issue be reproduced?

1. Launch the AirSim simulation on Ubuntu 20.04 and any environment under Unreal Engine 4.25.4.
2. Launch AirSim ROS wrapper: `roslaunch airsim_ros_pkgs airsim_node.launch`
3. Monitor the GPU memory consumption using `nvidia-smi`

### Include full error message in text form
```
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Segmentation fault (core dumped)
```

yhabib29, 2022-04-06:
I found out a solution.
My Environment is packaged for Linux (No Editor), and it was launching in fullscreen. So I had to reduce the screen (Alt+Tab) to switch back to my terminal and launch AirSim ROS wrapper. When starting the ROS node, the GPU memory usage was increasing until it saturates.

The workaround is to launch the Game using `-windowed` argument, for instance:
```
./Blocks.sh -windowed

``` 
Now it works well !

closed 2022-04-06 by yhabib29

hoangvietdo, 2022-06-15:
@yhabib29  Could you get synced images across 2 images when using stereo camera?