microsoft/AirSim #4492: Ardupilot with airsim: Error while receiving rotor control data

Kevinlibaba, 2022-04-21:
## Bug report 

- AirSim Version/#commit:Binary 1.7.0
- UE/Unity version:N/A
- autopilot version:master
- OS Version:Ubuntu 18.04

### What's the issue you encountered?

When I run the binary with ArduCopter's setting, an error was occurred as follows:
```
[2022.04.21-06.56.16:031][  0]LogStaticMesh: Allocated 512x387x128 distance field atlas = 48.4Mb, with 220 objects containing 33.5Mb backing data
Error while receiving rotor control data - ErrorNo: -1
[2022.04.21-06.56.16:108][  0]LogTemp: Error while receiving rotor control data - ErrorNo: -1
Error while receiving rotor control data - ErrorNo: -1
[2022.04.21-06.56.16:208][  0]LogTemp: Error while receiving rotor control data - ErrorNo: -1
[2022.04.21-06.56.16:217][  0]LogContentStreaming: Texture pool size now 1000 MB
Error while receiving rotor control data - ErrorNo: -1
[2022.04.21-06.56.16:309][  0]LogTemp: Error while receiving rotor control data - ErrorNo: -1
[2022.04.21-06.56.16:407][  0]LogVulkanRHI: Creating new VK swapchain with present mode 0, format 44, color space 0, num images 3
Error while receiving rotor control data - ErrorNo: -1
[2022.04.21-06.56.16:410][  0]LogTemp: Error while receiving rotor control data - ErrorNo: -1
Error while receiving rotor control data - ErrorNo: -1
Error while receiving rotor control data - ErrorNo: -1
Error while receiving rotor control data - ErrorNo: -1
Error while receiving rotor control data - ErrorNo: -1
```

### How can the issue be reproduced?

1.Set the 'VehicleType' in settings to ArduCopter/ArduRover, and run airsim with binary. the error will be occurred.
2.If set the 'VehicleType' back to SimpleFlight", anything went to well.

### Include full error message in text form
see log [file.](https://gist.github.com/Kevinlibaba/25cb7cb6b74bb41336c3a8baa3cf880b)
### Settings

my setting.json is as follows:
```
{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002
    }
  }
}

```

**What's better than filing an issue? Filing a pull request :).**

samuelctabor, 2022-05-22:
Are you running both on Linux?  For using on Windows with Ardupilot on a separate machine or in WSL2 you'll need some settings changes.

Kevinlibaba, 2022-05-23:
> Are you running both on Linux? For using on Windows with Ardupilot on a separate machine or in WSL2 you'll need some settings changes.

Yes, both are running on Ubuntu 18.04

talha-opteran, 2022-06-11:
I believe the AirSim binary is waiting for a response from Ardupilot. If you run Ardupilot, it should resume. Usually the execution order is Ardupilot and then AirSim.

dimikout3, 2022-09-29:
I have recently encountered the same problem. A quick fix is changing the UdpPort and ControlPort

{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 5760,
      "ControlPort": 5760
    }
  }
}

However, when the AirSim is started the quadcopter just increase its vertical speed constantly, so you can not really control it.

WangLouis, 2024-07-29:
Test environment:
Windows 11 Pro build 22000.708,
running Ubuntu 22.04LTS under WSL2. 
Installed Unreal Engine v4.27.2 via Epic Games Launcher.
Complied AirSim under Visual Studio 2022 (Community) , and ArduPilot in WSL2 with waf.

same issue
1.Set the 'VehicleType' in settings to ArduCopter, and run airsim with binary. the error will be occurred.
2.If set the 'VehicleType' back to SimpleFlight", anything went to well.

Has anyone solved the problem? Thanks

crose72, 2024-07-29:
@WangLouis @Kevinlibaba try this: https://discuss.ardupilot.org/t/gsoc-2019-airsim-simulator-support-for-ardupilot-sitl-part-ii/46395/29?u=crose4

I had this problem for 2 years which kept me from using AirSim with WSL2.  Getting the ip address of WSL2  and windows like the post suggests is what solved it for me. Hope it helps!

WangLouis, 2024-07-31:
@crose72 
thank you for your help~ but i can't solved it.

1.When i use ip address of WSL2 , It is error
https://www.youtube.com/watch?v=YJyt5aou1rI

2.'VehicleType' back to SimpleFlight", anything went to well.
https://www.youtube.com/watch?v=MqZPbHv2HoQ

3.Set the 'VehicleType' in settings to ArduCopter, the error will be occurred.
https://www.youtube.com/watch?v=G8It1qr6jrk

Should AirSim be executed first or WSL2 SITL be executed first? Thank you

WangLouis, 2024-08-06:
@crose72 

That worked, I got it , thank you so much!

“LocalHostIp”: “192.168.0.85”, ← this is IP of your main Windows
“UdpIp”: “172.26.240.75”, ← this is IP of your Linux(Ubuntu) under WSL2.
AirSim be executed first.
WSL2 Ardupilot SITL be executed second.

crose72, 2024-08-06:
> @crose72 
> 
> That worked, I got it , thank you so much!
> 
> “LocalHostIp”: “192.168.0.85”, ← this is IP of your main Windows
> “UdpIp”: “172.26.240.75”, ← this is IP of your Linux(Ubuntu) under WSL2.
> AirSim be executed first.
> WSL2 Ardupilot SITL be executed second.

Sorry I didn't get back to you earlier. I'm glad you've got it now!

valeriavalery, 2024-08-26:
> I have recently encountered the same problem. A quick fix is changing the UdpPort and ControlPort
>
> ```
> {
>   "SettingsVersion": 1.2,
>   "LogMessagesVisible": true,
>   "SimMode": "Multirotor", [...]

Hi! I have the same problem... I changed the ports but after that the quadcopter started to increase its vertical speed... do you know how can I solve?