microsoft/AirSim #4494: LiDAR Pointcloud to 2D image

JuanshuB2, 2022-04-23:
## Question
### What's your question?
Is it possible to get the LiDAR debug points as an 2D image directly for AirSim? Like the one above but only the green points. whitout the objects?

[image]

### Include context on what you are trying to achieve
I want the 2D image from the LiDAR pointcloud not to do it manually.

### Include details of what you already did to find answers
The image could be get from the 3D point cloud transformation to 2D, but if AirSim already give us that projection, it can be useful to be able to delete the background in the image. I have figured out about subtracting an image with the green points and another without them, but as the points are visible in the world, I have no access to a image as the second one.

jonyMarino, 2022-05-02:
@JuanshuB2 that is not currently possible. Why do you need the 2D image of the points reached by the lidar?

JuanshuB2, 2022-05-04:
Thanks for your reply @jonyMarino. I need the 2D image because we want to do the overlapping camera-lidar. Now we are transforming the 3D pointcloud to 2D image but the overlap sometimes is not easy, so a ground truth like that would be great. However, I guess we can work with that image, even if we can not have both. Maybe some kind of image processing can be done to delete the green points.