microsoft/AirSim #4500: multirotor fly in straight line

Zzerr0r, 2022-04-27:
### Discussed in https://github.com/microsoft/AirSim/discussions/4497

<div type='discussions-op-text'>

<sup>Originally posted by **Zzerr0r** April 25, 2022</sup>
Hi, I am just trying to fly in a straight line.
I am using moveToPositionAsync(200,0,0,40) so i expected the drone to just fly 200 meters in x-direction and 0 in y- and z-direction. However the drone keeps dropping until it eventually crashes into the ground along its path.  I have also tried moveByVelocityAsync,  moveByVelocityZBodyFrameAsync and moveByVelocityBodyFrameAsync (whatever the difference between these 3 is --> is there no documentation or can i just not find it?) where i also set the z-velocity to 0, resulting in the drone crashing into the ground again (which is flat by the way). I tried changing the PhysicsEngineName setting in the settings.json to "ExternalPhysicsEngine" as suggested in another post (https://github.com/microsoft/AirSim/issues/2570), which resulted in the drone not moving at all anymore (as to be expected tbh). 
Any help is appreciated. Kinda losing my mind over something this basic. Thanks in advance!</div>

closed 2022-05-09 by jonyMarino

quikssb, 2023-05-14:
did you find a solution? @Zzerr0r

Zzerr0r, 2023-05-14:
Not really, I found that the physics model of the drone isn't really made for flying at high speeds so I went over to just teleporting the drone in small intervals, which produces recording footage indistuingishable from continuous flight