microsoft/AirSim #4502: client.getDistanceSensorData() not working in ComputerVision mode

Zzerr0r, 2022-04-27:
## Bug report 

- AirSim Version/#commit: 1.7.0-windows
- UE/Unity version: 4.26
- autopilot version: N/A
- OS Version: windows 10

### What's the issue you encountered?

client = airsim.VehicleClient()
client.getDistanceSensorData()
´´´ 
leads to     
´´´
raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getDistanceSensorData' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for '' is not available. This could either because this is simulation-only API or this vehicle does not exist.
´´´

### Settings
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "Vehicles": {
    "Drone": {
      "VehicleType": "ComputerVision",
      "X": 0, "Y": 0, "Z": 0,
      "Sensors": {
        "Distance": {
          "SensorType": 5,
          "Enabled": true,
          "MinDistance": 0.2,
          "MaxDistance": 40,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "Yaw": 0,
          "Pitch": 0,
          "Roll": 0,
          "DrawDebugPoints": false
        }
      }
    }
  },
  "DefaultSensors": {
   "Distance": {
        "SensorType": 5,
        "Enabled" : true,
        "MinDistance": 0.2,
        "MaxDistance": 40,
        "X": 0, "Y": 0, "Z": -1,
        "Yaw": 0, "Pitch": 0, "Roll": 0,
        "DrawDebugPoints": false
    }
  }
}

alonfaraj, 2022-04-27:
Currently, computer vision doesn't have sensors like gps, distance etc. 
That's the error message you got regarding the simulation only api.

Zzerr0r, 2022-04-27:
Oh, that's unfortunate, thanks for the reply tho!

closed 2022-04-27 by Zzerr0r