microsoft/AirSim #4507: Error while building AirSim

huytd-pi, 2022-04-29:
## Bug report 

- AirSim Version/#commit: v1.7.0
- UE/Unity version: 4.25
- autopilot version: PX4
- OS Version:Ubuntu 18.04

I do AirSim build on linux, processes I follow steps like this path: https://microsoft.github.io/AirSim/build_linux/
I followed the steps one by one until the AirSim build step got an error.
Every time I run the build, there is a different message like:
_make: *** [cmTC_9d0d3/fast] Error 2_
_make: *** [cmTC_fb769/fast] Error 2_

> huytd@huytd-HP:~/Project/Drone/AirSim/AirSim$ `./setup.sh`
> + set -e
> +++ dirname ./setup.sh
> ++ cd .
> ++ pwd
> + SCRIPT_DIR=/home/huytd/Project/Drone/AirSim/AirSim
> + pushd /home/huytd/Project/Drone/AirSim/AirSim
> + downloadHighPolySuv=true
> + MIN_CMAKE_VERSION=3.10.0
> ++ uname
> + '[' Linux == Darwin ']'
> + DEBUG=false
> + [[ 0 -gt 0 ]]
> ++ uname
> + '[' Linux == Darwin ']'
> + sudo apt-get update
> [sudo] password for huytd: 
> Hit:1 http://vn.archive.ubuntu.com/ubuntu bionic InRelease
> Hit:2 http://vn.archive.ubuntu.com/ubuntu bionic-updates InRelease             
> Hit:3 http://vn.archive.ubuntu.com/ubuntu bionic-backports InRelease           
> Hit:4 https://download.docker.com/linux/ubuntu bionic InRelease                
> Hit:5 https://dl.google.com/linux/chrome/deb stable InRelease                  
> Hit:6 https://linux.teamviewer.com/deb stable InRelease                        
> Hit:7 https://download.mono-project.com/repo/ubuntu stable-bionic InRelease    
> Hit:8 https://packages.microsoft.com/repos/ms-teams stable InRelease           
> Hit:9 http://ppa.launchpad.net/ubuntu-mozilla-security/ppa/ubuntu focal InRelease
> Hit:10 http://security.ubuntu.com/ubuntu bionic-security InRelease             
> Hit:11 http://packages.ros.org/ros/ubuntu bionic InRelease                     
> Hit:12 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic InRelease 
> Hit:13 http://ppa.launchpad.net/bablu-boy/nutty/ubuntu bionic InRelease        
> Hit:14 http://ppa.launchpad.net/bamboo-engine/ibus-bamboo/ubuntu bionic InRelease
> Hit:15 https://librealsense.intel.com/Debian/apt-repo bionic InRelease         
> Hit:16 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu bionic InRelease         
> Hit:17 http://ppa.launchpad.net/elementary-os/stable/ubuntu bionic InRelease   
> Hit:18 http://ppa.launchpad.net/maateen/battery-monitor/ubuntu bionic InRelease
> Hit:19 http://ppa.launchpad.net/obsproject/obs-studio/ubuntu bionic InRelease  
> Hit:20 http://ppa.launchpad.net/saiarcot895/flightgear/ubuntu bionic InRelease 
> Hit:21 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu bionic InRelease
> Hit:22 http://ppa.launchpad.net/ubuntugis/ppa/ubuntu bionic InRelease          
> Reading package lists... Done 
> + sudo apt-get -y install --no-install-recommends lsb-release rsync software-properties-common wget libvulkan1 vulkan-utils
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> lsb-release is already the newest version (9.20170808ubuntu1).
> rsync is already the newest version (3.1.2-2.1ubuntu1.4).
> software-properties-common is already the newest version (0.96.24.32.18).
> wget is already the newest version (1.19.4-1ubuntu2.2).
> libvulkan1 is already the newest version (1.1.70+dfsg1-1ubuntu0.18.04.1).
> vulkan-utils is already the newest version (1.1.70+dfsg1-1ubuntu0.18.04.1).
> 0 upgraded, 0 newly installed, 0 to remove and 66 not upgraded.
> ++ lsb_release -rs
> ++ cut -d. -f1
> + VERSION=18
> + '[' 18 -lt 17 ']'
> + sudo apt-get install -y clang-8 clang++-8 libc++-8-dev libc++abi-8-dev
> Reading package lists... Done
[... 205 lines cut ...]
>     clang: error: linker command failed with exit code 1 (use -v to see invocation)
>     CMakeFiles/cmTC_c8dc5.dir/build.make:99: recipe for target 'cmTC_c8dc5' failed
>     make[1]: *** [cmTC_c8dc5] Error 1
>     make[1]: Leaving directory '/home/huytd/Project/Drone/AirSim/AirSim/build_release/CMakeFiles/CMakeTmp'
>     Makefile:127: recipe for target 'cmTC_c8dc5/fast' failed
>     make: *** [cmTC_c8dc5/fast] Error 2
>     
>     
> 
>   
> 
>   CMake will not be able to correctly generate this project.
> Call Stack (most recent call first):
>   CMakeLists.txt:2 (project)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/huytd/Project/Drone/AirSim/AirSim/build_release/CMakeFiles/CMakeOutput.log".
> See also "/home/huytd/Project/Drone/AirSim/AirSim/build_release/CMakeFiles/CMakeError.log".
> + popd
> ~/Project/Drone/AirSim/AirSim ~/Project/Drone/AirSim/AirSim
> + rm -r build_release
> + exit 1

**_I really appreciate everyone helping me fix this error!_**

jonyMarino, 2022-05-09:
Hi and welcome @huytd-pi! See if [this post](https://stackoverflow.com/questions/2086072/linking-using-g-fails-searching-for-lstdc) can help you.

jclee0109, 2022-11-11:
Hello @huytd-pi! I just met the same problem as you. I thought that clang compiler didn't work well in our environment (I don't have any idea about the specific reason..) Anyway, so I changed the compiler as gcc using `./build.sh --gcc` and it works.

I hope you can fix your problem as well. Thanks.

rishi1906, 2023-11-25:
@jclee0109 Thanks alot -gcc resolved the issue