microsoft/AirSim #4517: water foam and wave effects of the sea do not apply when in airsim mode

bene-sim, 2022-05-09:
## Bug report 
- AirSim Version/#commit: v1.7
- UE version: 4.26.2
- autopilot version: computer vision
- OS Version: WIN10

### What's the issue you encountered?
water foam and wave effects of the sea do not apply when in airsim mode.

It is a sea environment purchased from the marketplace of Unreal Engine, and as shown in the image below, the effects of the sea are well applied in the basic gamemode, but the effects are not applied in the airsim mode. (The Sea Asset is made in blue print.)

Various effects work well when it is not in the air sim mode, but the effects disappear when the air sim mode is executed.
does anyone know how to solve same problem?

-﻿ThirdPersonGameMode(basic gamemode)
[image]

-AirSimGameMode
[image]

In addition, if the altitude rises above a certain level, a strange plane is created as shown in the figure below.
[image]

### Settings
{
  "SettingsVersion": 1.7,
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 3840,
        "Height": 2160,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      }
    ]
  },
  "SimMode": "ComputerVision"
}

### How can the issue be reproduced?
Set the game mode to the airsim mode in the sea environment and then start the game.

zimmy87, 2022-05-12:
Hi @bene-sim, from looking at our AirSimGameMode, there's nothing that jumps out to me that could be causing the rendering of an effect like this to get disabled. The ThirdPersonGameMode you were using before isn't an engine class, so it may be doing some project-specific initialization of classes or variables that the wave effect depends upon. I would check the implementation of that class to see if it's doing anything that needs to be copied over to AirSimGameMode. Another possibility is that the wave effect isn't correctly configured to render in a cinematographic camera. We switched over to the CineCameraComponent in #3949, you try with a version that doesn't have that change applied (v1.6.0 or lower) to see if this is due to which camera we're using.

bene-sim, 2022-05-18:
UE4.25 - airsim1.6
UE4.26 - airsim1.6
UE4.26 - airsim1.7
The list above does not apply.
This is a new fact that not only the "airsim"mode but also the "functionaltestgame" mode has the same problem.

bene-sim, 2022-05-18:
I think this problem is originally a problem for this Asset.
I'm sorry for the confusion.
[image]

jonyMarino, 2022-05-19:
@bene-sim Thanks for these new reports. Now it doesn't seem to be an AirSim issue. You would need to modify other game modes (including AirSim game mode) to get that plugin to work with them.