microsoft/AirSim #4524: PX4 HITL + External Offboard Controller + AirSim

xubeyb, 2022-05-16:
Hi everyone and especially Mr. Chris Lovett (@lovettchris),

What is the best way to include my external offboard controller in my simulation environment? 

I am trying to setup a HITL simulation environment for my drone using AirSim. I use a companion computer connected to autopilot through a serial port (a mavlink channel) that takes care of offboard control and some other stuff. I have been using AirSim since a long time and am quite familiar with the sources. I know that AirSim is ready for PX4 HITL and that its configuration is straight forward through settings.json (UseSerial and etc.). Here are my questions:

**Question 1:** 
What I have tried to do first was just set AirSim as HITL in settings.json as described in the docs to connect my autopilot to the simulator through USB (ttyACMx) and connect my offboard controller to my autopilot through a serial port as it is in the real system. This caused some instabilities in flights, actually sometimes it led to a meaningless freefall after some flight time (3-5 minutes). I actually couldn't find a reason for this either in airSim or px4 logs. But I think airSim is somehow and somewhat interfering with offboard controller mavlink messages. Can we discuss this? What do you think is wrong with my setup?

**Question 2:** 
After failing my first solution, I thought of connecting the autopilot to my offboard controller through airSim and implemented an offboard node in MavLinkMultirotorApi.hpp and have tried to "route" offboard controller messages to my pixhawk and pixhawk messages to offboard controller. But I think I am misunderstanding the code. Do you think I am on the right path?

Thanks

Ubuntu 20.04 LTS
AirSim v1.7.0 (Built from minorly customized source)
UE4.25

`{

    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Wind": {
        "X": 0,
        "Y": 0,
        "Z": 0
    },
    "PawnPaths": {
        "DefaultQuadrotor": {
            "PawnBP": "Class'/Game/Blueprints/BP_FlyingPawn.BP_FlyingPawn_C'"
        }
    },
    "DefaultSensors": {
        "Barometer": {
            "SensorType": 1,
            "Enabled": true
        },
        "Imu": {
            "SensorType": 2,
            "Enabled": true
        },
        "Gps": {
            "SensorType": 3,
            "Enabled": true
        },
        "Magnetometer": {
            "SensorType": 4,
            "Enabled": true
        },
        "Lidar": {
            "SensorType": 6,
            "Enabled": false,
            "NumberOfChannels": 4,
            "PointsPerSecond": 10000
        }
    },
    "OriginGeopoint": {
        "Latitude": 39.9334,
        "Longitude": 32.8597,
        "Altitude": 889
    },
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "Model": "MyCustomMultirotor",
            "SerialBaudRate": 921600,
            "SerialPort": "*",
            "UseSerial": true,
            "UseTcp": false,
            "ControlIP": "127.0.0.1",
            "ControlPortLocal": 14580,
            "ControlPortRemote": 14540,
            "UdpIp": "",
            "UdpPort": 14540,
            "TcpPort": 4560,
            "VehicleCompID": 1,
            "VehicleSysID": 142,
            "LocalHostIp": "127.0.0.1",
            "params": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            }
        }
    }
}`

lovettchris, 2022-05-16:
I have a couple ideas:
1) Have you tried it without the Lidar sensor - the Lidar is expensive in CPU and this could be causing AirSim to run too slowly, and send messages to the Pixhawk too slowly causing the simulation to fall behind.
2) But falling behind is solved by running PX4 in "lockstep" mode.  See [Lockstep Mode](https://github.com/microsoft/AirSim/blob/12d9c11b46fefbea275077621b51360046203ac5/docs/px4_lockstep.md) and follow those steps to ensure the pixhawk is in lockstep mode then the simulation should work well even when it is running slowly.

xubeyb, 2022-05-16:
@lovettchris thanks, I'll give lockstep a try and will remove lidar as I do not need it anymore. Will update here

xubeyb, 2022-05-17:
Lockstep is for tcp connection i.e SITL right? I am trying to achieve this with a serial connection

xubeyb, 2022-05-17:
By the way this happens after like 3.00-3.15 minutes. The video is shortened for size

lovettchris, 2022-05-17:
Thanks for the video.  Yes, your system is more than capable of running AirSim.   It does appear like lockstep got stuck.  It has the classic symptoms.  I'm recalling that a normal pixhawk firmware build (designed for flying real drones) does NOT have lockstep enabled by default which means you will have to build the firmware with lockstep enabled.  But can you try another test before you do that.  Can you try and disable lockstep by explicitly setting it to false in the AirSim settings.json file?  Lockstep will only work if both sides have it enabled.  If you tell AirSim not to use lockstep then it won't get in a "stuck state" waiting for a lockstep reply that is never coming.

xubeyb, 2022-05-17:
I gave it a shot. Unfortunately the result was same. I'll give it another try tomorrow

xubeyb, 2022-05-17:
What I don't understand is; ain't lockstep an SITL feature? Can it be activated in HITL also?

lovettchris, 2022-05-17:
Right, you can see [in the code ](https://github.com/PX4/PX4-Autopilot/search?q=lockstep) that it is only enabled for SITL, here: 

[image]

Now you are using HITL, and so I'm wondering if you could build the HITL firmware with lockstep enabled, things might be a lot more predictable and you can run the simulation as slow as you want (e.g. with Lidar and anything else you need) even step through the code in the debugger and everything just works.

I'm surprised it gets stuck when lockstep is disabled.  Can you confirm you set the lockstep to false with `"LockStep": false,"`, rather than removing lockstep setting?

xubeyb, 2022-05-18:
I'll try these on Monday and will update you. Thanks for keeping up

xubeyb, 2022-05-30:
I tried building PX4 (HITL) with lockstep enabled but what I have found out that the lockstep_scheduler implementation is only available for posix. Therefore it has to be ported for nuttX. So, currently, there is no implementation of lockstep in PX4 except for SITL.

xubeyb, 2022-05-30:
Setting lockstep to "false" in settings.json doesn't help. After 3 minutes of normal flight. Same thing happens :/

xubeyb, 2022-05-31:
I have set the OriginGeopoint lat and lons and px4 LPE_LAT/LON to Redmond. Same thing happens... I wonder if I can trace anything through AirSim logs? When I check the contents of "Saved" folder, everything seems to be normal. Only that px4 logs say ACCEL #0 TIMEOUT and BARO #0 TIMEOUT.

xubeyb, 2022-05-31:
@Jaeyoung-Lim any ideas about this?

xubeyb, 2022-06-08:
Hi, did anyone had the chance to reach any conclusion with this?

xubeyb, 2022-09-26:
@Jaeyoung-Lim @lovettchris  is there any news about this?