microsoft/AirSim #4535: Cannot open Blocks.uproject with UE4.27

WarrenSkywalker, 2022-05-20:
## Question
### What's your question?

I followed the instruction here:https://microsoft.github.io/AirSim/unreal_blocks/, however, the UE4 cannot open the default Blocks environment in the AirSim. I found the version of the engine that made the Blocks.uprojcet is 4.27, which is the same as my UE version, it's wield that UE recognized the project came from an engine of different version.
### Include context on what you are trying to achieve

#### Context details

Ubuntu 18.04LTS, AirSim master branch from its github, Unreal 4.27

### Include details of what you already did to find answers
I've tried convert it in place, but met a compile error:
Building would modify the following engine files:
xxxxx/UE4Editor.modules
.....

LongruiDong, 2022-05-23:
Hi .
I am also meeting this error, have you solved yet?

WarrenSkywalker, 2022-05-23:
not yet, hope sb could help us

Davidsastresas, 2022-05-23:
I am having the same issue all of a sudden!

andrewjong, 2022-05-24:
Hi all, I managed to open Blocks.uproject on my machine. UE4.27 on Ubuntu 18.04.

1. Cloned UnrealEngine to ~/UnrealEngine. Cloned AirSim to ~/AirSim
2. Cleaned the UBlocks project `cd ~/AirSim/Unreal/Environments/Blocks && ./clean.sh`
3. Installed [mono-complete for Ubuntu 18.04](https://www.mono-project.com/download/stable/#download-lin)
4. Made some symlinks: `ln -s /usr/bin/mono /bin/mono && ln -s /usr/lib/mono /lib/mono`
5. Ran the following command to rebuild the Blocks project: ```~/UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/bin/mono ~/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project=/home/andrew/AirSim/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress```
6. Started UE4Editor `~/UnrealEngine/Engine/Binaries/Linux/UE4Editor`
7. Clicked More > Browse > `/home/andrew/AirSim/Unreal/Environments/Blocks/Blocks.uproject`
8. Error shows up "This project was made with a different version of the Unreal Engine.".   Click "More Options" > "Skip conversion"
9. Project opens, whala! Unfortunately you always have to click "Skip conversion" to open it. If anyone knows a better way, please feel free to share. Thanks!

zimmy87, 2022-05-24:
Thanks for looking into this and providing a solution @andrewjong! Another solution I found is to follow the instructions in this [forum post](https://forums.unrealengine.com/t/solved-upgrade-c-project-from-ue-4-24-to-4-25-under-linux/144950/11). Basically, this is happening because of an outdated BlocksEditor shared library left over from a previous build. I'll try and look into what specific build output needs to be cleaned up to avoid this, but in the meantime, everyone should be unblocked by either @andrewjong's solution or the forum post I linked to.

09zs19, 2022-05-24:
I have been running into the same issue for about a week or so and the solution from @andrewjong worked perfectly for me.  Thank you for the help!

Davidsastresas, 2022-05-28:
Thank you so much @andrewjong, your solution worked for me! Thanks thanks thanks!

aliadeeba98, 2022-06-02:
> Hi all, I managed to open Blocks.uproject on my machine. UE4.27 on Ubuntu 18.04.
> 
> 1. Cloned UnrealEngine to `~/UnrealEngine`. Cloned AirSim to `~/AirSim`
> 2. Cleaned the UBlocks project `cd ~/AirSim/Unreal/Environments/Blocks && ./clean.sh`
> 3. Installed [mono-complete for Ubuntu 18.04](https://www.mono-project.com/download/stable/#download-lin), which lets you run .exe files on Linux
> 4. Made some symlinks: `sudo ln -s /usr/bin/mono /bin/mono && sudo ln -s /usr/lib/mono /lib/mono`
> 5. Ran the following command to rebuild the Blocks project: `~/UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/bin/mono ~/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project=/home/andrew/AirSim/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress`
> 6. Started UE4Editor from command line: `~/UnrealEngine/Engine/Binaries/Linux/UE4Editor`
> 7. Clicked More > Browse > `/home/andrew/AirSim/Unreal/Environments/Blocks/Blocks.uproject`
> 8. Error shows up "This project was made with a different version of the Unreal Engine.".   Click "More Options" > "Skip conversion"
> 9. Project opens, whala! Unfortunately you always have to click "Skip conversion" to open it. If anyone knows a better way, please feel free to share. Thanks!

But sir, after executing the command wine UnrealBuildTool.exe Development Linux -Project=/home/andrew/AirSim/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress,  I am getting the following error:
wine: Unhandled page fault on read access to 0000000000000004 at address 000000007BCDCAE4 (thread 0019), starting debugger...
000f:err:service:process_send_command service protocol error - failed to write pipe!
Using 'git status' to determine working set for adaptive non-unity build (Z:\home\adeeba\UnrealEngine).
Invalidating makefile for BlocksEditor (Paper2D.uplugin has been added)
ERROR: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.

And I'm stuck here! Please help me out in this regard.

Thank you.

andrewjong, 2022-06-02:
@aliadeeba98 , based on the backslashes in your path, it looks like you're on Windows. The instructions I wrote are for Linux. However perhaps it will still work if you simply run `~/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Windows -Project=/home/andrew/AirSim/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress`? Not entirely sure because I'm not on Windows myself.

aliadeeba98, 2022-06-03:
> @aliadeeba98 , based on the backslashes in your path, it looks like you're on Windows. The instructions I wrote are for Linux. However perhaps it will still work if you simply run [...]

@andrewjong I'm working on Ubuntu 20.04 LTS and for executing the UnrealBuildTool.exe file in Linux I am taking the help of Wine software.

andrewjong, 2022-06-03:
`mono` should let you run the `.exe` in Linux

jacobsayono, 2022-06-29:
@andrewjong hi, your solution worked to compile Blocks, but in the end, i would still get the error:
Engine modules are out of date, and cannot be compiled while the engine is running. Please build through your IDE.
do you think it is because i am using Ubuntu 20.04 LTS?

tanyaapaul, 2022-10-10:
same error on my Ubuntu 20.04 LTS any tips or individual that managed to solve this issue?

andrewjong, 2023-01-18:
Since AirSim is now deprecated, I might suggest transitioning to Colosseum Sim, which is a fork of AirSim. They aim to carry on AirSim to Unreal Engine 5 (and apparently theirs already works for UE5).

panyaoqiang, 2023-06-15:
> I have also encountered this problem. Have you resolved it

panyaoqiang, 2023-06-19:
您好，您的邮件耀强已经收到，辛苦了。

cheslee-z, 2023-07-24:
> @aliadeeba98 , based on the backslashes in your path, it looks like you're on Windows. The instructions I wrote are for Linux. However perhaps it will still work if you simply run [...]
@KadriyeNur, I also encounter your problem when using attentioned method, everything seems to be OK by just disabling the plugin. By the way, my PC is ubuntu 20.04

panyaoqiang, 2023-07-24:
您好，您的邮件耀强已经收到，辛苦了。

cheslee-z, 2023-07-24:
> Thanks for looking into this and providing a solution @andrewjong! Another solution I found is to follow the instructions in this [forum post](https://forums.unrealengine.com/t/solved-upgrade-c-project-from-ue-4-24-to-4-25-under-linux/144950/11) (I suspect this essentially doing the same thing as step 5 in his solution, just with a make command instead of using the UnrealBuildTool). Basically, this is happening because of an outdated BlocksEditor shared library left over from a previous build. I'll try and look into what specific build output needs to be cleaned up to avoid this, but in the meantime, everyone should be unblocked by either @andrewjong's solution or the forum post I linked to.
> 
> Here's the solution offered in the previously linked forum post:
> 
> ```
> /home//UE/UnrealEngine/GenerateProjectFiles.sh /home/projectName/projectName.uproject -game
> cd /home/projectName/
> make projectNameEditor
> ```

@KadriyeNur, I also encounter your problem when using attentioned method, everything seems to be OK by just disabling the plugin. By the way, my PC is ubuntu 20.04

tbh123456, 2024-01-30:
> 您好，您的邮件耀强已经收到，辛苦了
I have meet the same question ,do you know how to disable the SteamVR?

panyaoqiang, 2024-01-30:
您好，您的邮件耀强已经收到，辛苦了。

IaroslavS, 2024-02-15:
> Hi all, I managed to open Blocks.uproject on my machine. UE4.27 on Ubuntu 18.04.
>
>     1. Cloned UnrealEngine to `~/UnrealEngine`. Cloned AirSim to `~/AirSim`
>
>     2. Cleaned the UBlocks project `cd [...]

I have Unreak Engine 5.2 and I don't have UnrealBuildTool, I found only UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool/UnrealBuildTool.
When I've run 
```
UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/bin/mono UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool/UnrealBuildTool Development Linux -Project=/home/user/workspace/Colosseum/Unreal/Environments/Blocks/Blocks.uproject -TargetType=Editor -Progress
```
An error occurs:
```
Cannot open assembly 'UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool/UnrealBuildTool': File does not contain a valid CIL image.
```

panyaoqiang, 2024-02-15:
您好，您的邮件耀强已经收到，辛苦了。

nvhungv2k, 2024-07-02:
> Hi all, I managed to open Blocks.uproject on my machine. UE4.27 on Ubuntu 18.04.
>
>     1. Cloned UnrealEngine to `~/UnrealEngine`. Cloned AirSim to `~/AirSim`
>
>     2. Cleaned the UBlocks project `cd [...]

Thanks @andrewjong Your way worked for me.
My workspace: Ubuntu 20.04LTS, Airsim branch master, Unreal Engine 4.27

panyaoqiang, 2024-07-02:
您好，您的邮件耀强已经收到，辛苦了。

rahmarzougui, 2024-09-19:
> 6\. ~/UnrealEngine/Engine/Binaries/Linux/UE4Editor

Thank you this worked for me!