microsoft/AirSim #4538: How to rescale DepthVis max distance?

HSC-1, 2022-05-23:
## Question
### What's your question?

I modified the blueprint as shown in the picture below to make it more subdivided by reducing the size of the depth.
[image]

So it worked fine in the editor as intended.

[image]

But when I received the data from Python, it didn't work as intended.
[image]

python code

```
raw = client.simGetImages([airsim.ImageRequest(0,airsim.ImageType.DepthVis,True,False)])
depth = np.array(raw[0].image_data_float,dtype=np.float)
depth = np.reshape(depth,(raw_img[0].height,raw_img[0].width))
```

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

jonyMarino, 2022-06-02:
Hi @HSC-1! Which application are you using to see the image? Can you use PfmPad? Also, I'm experiencing a similar issue but saving a png image (pixels as float false)

HSC-1, 2022-06-13:
Thanks for the reply.
The same thing happens whether pixels as float is True or False. I also tried changing the DepthScaleFactor as per your recommendation, but got a similar problem. It seems to be a problem that is happening because I am working on UE5. Thank you for caring me.