microsoft/AirSim #4555: UE5 - AirSim prevents low-level details crowd and traffic from being drawned

czero69, 2022-06-01:
### About

With few tweaks, I have added AirSim plugin to UE5 cityscape demo. What I want is to use computer vision mode to save buffers. For some reason, when using AirSimGameMode, it disables drawing for low-level mass ai objects. These objects are still tracked and existing (green squares), just drawing is disabled.

I want to make low-level mass ai objects to be drawn. Do you have some ideas what can be the cause and what should I check in the first place?

### How Has This Been Tested?

Debugging with `MassTraffic.DebugVisualization 1` I concluded that higher level details mass ai objects are drawn as desired (all orange/red squares), but lower level-detail mass ai objects are not drawn at all (green squares).

Compiling AirSim to UE5 was quite straightforward. I just disabled not compatible code in car pawns (I only need computer vision mode). Please let me know, and I will paste the link to my fork here if someone wants to recreate the issue. Also, my settings.json is pretty standard.

### Screenshots (if appropriate):

[image]

[image]

czero69, 2022-06-01:
Ok Solved! 

The issue is related to origin position and/or calling GetWorld()->SetNewWorldOrigin(...) in AirSim. As a fast workaround, commenting SetNewWorldOrigin and setting spawning location same as City Sample Demo (not 0,0,0) solves the issue.

[image]

closed 2022-06-01 by czero69

leielyq, 2022-06-02:
Compiling AirSim to UE5 was quite straightforward. 

Share it, thank you

czero69, 2022-06-02:
here you are:

https://github.com/czero69/AirSim/tree/ue5-cv

Let me know if you are interested in City Sample Demo + Airsim. I plan to push additional fixes for that demo later.