microsoft/AirSim #4569: [Ros1] airsim_ros_wrapper outputs wrong time sync for odom_local_ned and stereo camera

hoangvietdo, 2022-06-15:
## Bug report 

- AirSim Version/#commit: Lastest
- UE/Unity version: 4.27
- autopilot version:
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

The timestamp of odom_local_ned topic and stereo camera topic is weird. In addition, there is a significant delay between the left and right camera (around 0.02 seconds which can be considered as significant for the stereo VINS algorithms). As shown in the figure, the publish rate of odom_local_ned and IMU are similar (~100 Hz), however, the actual recorded data illustrates that some data are lost in odom_local_ned topic.

[image]
[image]
[image]

### Settings

### How can the issue be reproduced?

1. This issue is produced using the default code of airism_ros_wrapper (ros1) from the latest branch.
2. The settings.json contains the basic settings for the stereo camera.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-06-22:
Hi @hoangvietdo ! It is posible that [this PR](https://github.com/microsoft/AirSim/pull/4559) can help reduce time differences. Let me know if that is the case.

hoangvietdo, 2022-06-24:
Hi @jonyMarino  , thanks so much for your help, unfortunately, I am not willing to switch to ros2 at the moment. 
I am trying to make airsim_ros_wrapper of ros2 be identical to ros1. However, there is some concept of ros2  (e.g., callBackGroup) that I could not find any alternative in ros1. Any help?

hoangvietdo, 2022-06-24:
In addition, I tested the PR and checked the recorded data. It seems like ros2 has the worst performance compared to ros1. It not only doesn't solve the time delay between 2 images, but it also makes the timestamp of IMU mixed up.

AirSim Version/#commit: #4559 
UE/Unity version: 4.27
autopilot version:
OS Version: Ros2 - Ubuntu 20.04 (Foxy)
[image]

closed 2022-08-08 by hoangvietdo