microsoft/AirSim #4570: AirSim failed to build with ""error C2065: 'path': undeclared identifier" with MSVC on Windows arm64ec

YangYang129, 2022-06-15:
AirSim failed to build with ""error C2065: 'path': undeclared identifier" with MSVC on Windows arm64ec. It can reproduce on latest version on master branch. Could you please help look at this issue?

**Repro steps:**
1. set VSCMD_SKIP_SENDTELEMETRY=1 & "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\Tools\VsDevCmd.bat" -host_arch=amd64 -arch=arm64
2. git clone https://github.com/microsoft/AirSim F:\AirSim
3. cd F:\AirSim
4. curl https://github.com/rpclib/rpclib/archive/v2.3.0.zip -O -L
5. 7z x  v2.3.0.zip -oF:\AirSim\external\rpclib -aoa
7. curl https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip -O -L
8. 7z x  eigen-3.3.7.zip -oF:\AirSim\AirLib\include -aoa
9. Change "#include "Eigen/Dense"" to "#include "eigen-3.3.7/Eigen/Dense"" in file VectorMath.hpp
10. cd cmake
11. mkdir build_arm64eca & cd build_arm64eca
12. cmake -G "Visual Studio 16 2019" -A ARM64EC -DCMAKE_SYSTEM_VERSION=10.0.22618.0 -DCMAKE_BUILD_TYPE=Release  ..
13. msbuild /m /p:Platform=ARM64EC /p:Configuration=Release AirSim.sln /t:Rebuild

**Error log:**
[AirSim_msbuild_error.log](https://github.com/microsoft/AirSim/files/8907065/AirSim_msbuild_error.log)

**Error info:**
```
17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(446,5): error C2065: 'path': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(446,10): error C2146: syntax error: missing ';' before identifier 'dirPath' [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(446,10): error C3861: 'dirPath': identifier not found [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,10): error C2065: 'directory_iterator': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,29): error C2146: syntax error: missing ';' before identifier 'next' [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,34): error C2065: 'dirPath': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,44): error C2065: 'end': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,49): error C2065: 'next': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,57): error C2065: 'end': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,62): error C2143: syntax error: missing ')' before '++' [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,62): error C2059: syntax error: ';' [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,64): error C2065: 'next': undeclared identifier [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(448,68): error C2059: syntax error: ')' [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
    17>F:\AirSim\MavLinkCom\MavLinkTest\Commands.cpp(449,9): error C2062: type 'auto' unexpected [F:\AirSim\cmake\build_arm64eca\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
```

jonyMarino, 2022-06-20:
Hi and welcome, @YangYang129! If you are using the latest master, you should use VS2022 for building.

YangYang129, 2022-07-29:
@jonyMarino I add msgpack in external\rpclib\rpclib-2.3.0\include\rpc, this project build successfully

closed 2022-07-29 by YangYang129