microsoft/AirSim #4578: ROS1 ambiguous camera tf frame_ids

lukas-utopiacompression, 2022-06-16:
## Bug report 

- AirSim Version/#commit: eda67e408 and v1.7
- UE/Unity version: 4.27.2
- autopilot version: n/a
- OS Version: Ubuntu 20.04.4

### What's the issue you encountered?

When multiple vehicles (see settings.json below) specify a camera with the same name (e.g. "img0") ROS1's tf frame_ids become ambiguous.

Related PR #4478 fixes the same issue but for ROS2.

Screenshot of rqt_tf_tree, when camera frame_ids are ambiguous. 
[image]

Screenshot of rqt_tf_tree, when camera frame_ids are distinct (e.g. by adding the vehicle_name to the frame_id):
[image]

### Settings

<details><summary>settings.json</summary>
<p>

```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ClockSpeed": 0.2,
  "SimMode": "Multirotor",
  "PhysicsEngineName": "ExternalPhysicsEngine",
  "ViewMode": "Manual",
  "DefaultSensors": {
    "imu_1": {
         "SensorType": 2,
         "Enabled" : true
    }
  },
  "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight",
      "Cameras": {
        "img0":{
          "CaptureSettings": [
            {
                "ImageType": 0,
                "Width": 1024,
                "Height": 1024,
                "FOV_Degrees": 90
            }
          ],
          "NoiseSettings": [
            {
            }
          ],
          "X": 0.5, "Y": 0.0, "Z": 0.0,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        }
      },
    "X": 0, "Y": 0, "Z": -1
    },
    "drone_2": {
      "VehicleType": "SimpleFlight",
      "Cameras": {
        "img0":{
          "CaptureSettings": [
            {
                "ImageType": 0,
                "Width": 1024,
                "Height": 1024,
                "FOV_Degrees": 90
            }
          ],
          "NoiseSettings": [
            {
            }
          ],
          "X": 0.5, "Y": 0.0, "Z": 0.0,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        }
      },
    "X": 2.0, "Y": 2.0, "Z": -1
    },
    "drone_3": {
      "VehicleType": "SimpleFlight",
      "Cameras": {
        "img0":{
          "CaptureSettings": [
            {
                "ImageType": 0,
                "Width": 1024,
                "Height": 1024,
                "FOV_Degrees": 90
            }
          ],
          "NoiseSettings": [
            {
            }
          ],
          "X": 0.5, "Y": 0.0, "Z": 0.0,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        }
      },
    "X": 2.0, "Y": 4.0, "Z": -1
    }
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1024,
        "Height": 1024,
        "FOV_Degrees": 80
      }
    ]
  },
  "SubWindows": [
    {"WindowID": 0, "VehicleName": "drone_1", "ImageType": 0, "CameraName": "img0", "Visible": true},
    {"WindowID": 1, "VehicleName": "drone_2", "ImageType": 0, "CameraName": "img0", "Visible": true},
    {"WindowID": 2, "VehicleName": "drone_3", "ImageType": 0, "CameraName": "img0", "Visible": true}
  ],
  "OriginGeopoint": {
      "Latitude": 0.0,
      "Longitude": 0.0,
      "Altitude": 0.0
    }
}

```

</p>
</details>

### How can the issue be reproduced?

1. Use above settings.json
2. open Blocks environment
3. `roslaunch airsim_ros_pkgs airsim_node.launch`
4. Open either rviz, or rqt with the rqt_tf_tree plugin

### Include full error message in text form
n/a

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-06-20:
Hi @lukas-utopiacompression! Thanks for submitting the PR. I'd like to know why you didn't upgrade to ROS2 when you ran into this issue.

lukas-utopiacompression, 2022-06-20:
Good morning @jonyMarino !

For two reasons:

1. Our project still requires ROS1. 
2. Fixing and creating a PR for the missing/ambiguous tf frame_ids seemed less work than transitioning our whole project to ROS1 (for now).

jonyMarino, 2022-06-20:
Thanks for the reply, @lukas-utopiacompression. Keep in mind that you can use the Ros1-Ros2 Bridge to connect your project with ROS2 nodes.