microsoft/AirSim #4580: ERROR occurred when opening Blocks.uproject

deadseason, 2022-06-20:
## Bug report 

- AirSim Version/#commit: 1.6.0 linux
- UE/Unity version: 4.27.2
- autopilot version: 
- OS Version: Ubuntu Ubuntu 20.04.4 LTS

### What's the issue you encountered?

For the first time I install the UE4.27 and airsim 1.6, after being installed, I open the example project ' Blocks.uproject ' with UE, but it does not work.Error occurred and it said compile failed. Is it the problem of Chinese directory path?

```
无法编译该项目。要在空源代码访问中打开它吗？

Running /home/gpu/存储1/UE4/UE4-27/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="/home/gpu/存储1/UE4/AIRSIM/AirSim-1.6.0-linux/Unreal/Environments/Blocks 4.27 - 2/Blocks.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/??????1/UE4/UE4-27/Engine ~/??????1/UE4/UE4-27/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build (/home/gpu/??????1/UE4/UE4-27).
Creating makefile for BlocksEditor (no existing makefile)
WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
@progress push 5%
Creating makefile for UnrealHeaderTool (no existing makefile)
@progress push 5%
@progress pop
------- Build details --------
Using toolchain located at '~/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/gpu/??????1/UE4/UE4-27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : ~/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/home/gpu/??????1/UE4/AIRSIM/AirSim-1.6.0-linux/Unreal/Environments/Blocks 4.27 - 2/Blocks.uproject" "/home/gpu/??????1/UE4/AIRSIM/AirSim-1.6.0-linux/Unreal/Environments/Blocks 4.27 - 2/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -abslog="/home/gpu/??????1/UE4/UE4-27/Engine/Programs/UnrealBuildTool/Log_UHT.txt"
LogInit: Display: Loading text-based GConfig....
Reflection code generated for BlocksEditor in 2.5155869 seconds
@progress pop
------- Build details --------
Using toolchain located at '/home/gpu/??????1/UE4/UE4-27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/gpu/??????1/UE4/UE4-27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/gpu/??????1/UE4/UE4-27/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
ERROR: Building would modify the following engine files:
       
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Android/libUE4Editor-AndroidDeviceDetection.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Android/libUE4Editor-AndroidPlatformEditor.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Android/libUE4Editor-AndroidRuntimeSettings.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Android/libUE4Editor-AndroidTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Android/UE4Editor.modules
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/libUE4Editor-VulkanRHI.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/libUE4Editor-WorldBrowser.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxAArch64ClientTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxAArch64NoEditorTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxAArch64ServerTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxClientTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxNoEditorTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxServerTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/libUE4Editor-LinuxTargetPlatform.so
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/Linux/UE4Editor.modules
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/UE4Editor
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/UE4Editor.modules
       /home/gpu/??????1/UE4/UE4-27/Engine/Binaries/Linux/UE4Editor-Cmd
       /home/gpu/??????1/UE4/UE4-27/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Chaos/AABB.ispc.generated.dummy.h
       /home/gpu/??????1/UE4/UE4-27/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Chaos/AABB.ispc.generated.h

```
[image]

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2022-06-20:
Hi and welcome @deadseason! Maybe you are facing #4264. Can you try with master?

deadseason, 2022-06-23:
> Hi and welcome @deadseason! Maybe you are facing #4264. Can you try with master?

Still can not open it. The same error.

jonyMarino, 2022-06-24:
I think I've found your issue. Try with [this solution](https://github.com/microsoft/AirSim/issues/4535#issuecomment-1136152996)

deadseason, 2022-06-28:
> I think I've found your issue. Try with [this solution](https://github.com/microsoft/AirSim/issues/4535#issuecomment-1136152996)

I have tried, but it still did not work.The same issue.

jonyMarino, 2022-06-29:
Did you upgrade to 4.27 coming from a previous version of unreal engine?

deadseason, 2022-06-29:
> Did you upgrade to 4.27 coming from a previous version of unreal engine?

Yes, I used 4.26.2 before, and the problem occurred. So I deleted the old version and installed  the new 4.27 UE.

deadseason, 2022-06-30:
> Ok. So can you try with a new clone of the AirSim repository?

Error still existed. I have updated  Airsim to 1.7.0.

jonyMarino, 2022-06-30:
@deadseason 
No. Not to update AirSim. An alternative would be for you to delete the Blocks folder. Then with a git pull, you should establish it.

deadseason, 2022-07-01:
> @deadseason No. Not to update AirSim. An alternative would be for you to delete the Blocks folder. Then with a git pull, you should restablish it.

how to use 'git pull' to restablish it?

deadseason, 2022-07-04:
> @deadseason, sorry it is with git checkout Blocks inside the Environments folder. Then you need to call update_from_git.bat/sh

I just run the update_from_git.sh, and still can not open it.

jonyMarino, 2022-07-04:
From your original post I see you are using mono. You can try with the [other solution](https://github.com/microsoft/AirSim/issues/4535#issuecomment-1136095046) provided in the previous thread.

tanyaapaul, 2022-10-05:
Same error , anyone managed to solve it ?

zhjc2014, 2022-11-11:
same error