microsoft/AirSim #4588: Problems using ExternalPhysicsEngine parameter in settings.json

firemount, 2022-06-22:
## Question
### What's your question?
Should Airsim open tcp port 4560 if "PhysicsEngineName" is set to ExternalPhysicsEngine in settings.json
### Include context on what you are trying to achieve
Hi I am trying out the ExternalPhysicsEngine parameter in settings.json to be able to use inject a FDM of my drone using PX4 with gazebo_iris and QGroungControl. Everything worked eventually but there is something I quite don't get. Following the manual https://microsoft.github.io/AirSim/gazebo_drone/ that says:

"First run the AirSim simulator and your Gazebo model and then execute this from your AirSim root folder:

cd GazeboDrone/build
./GazeboDrone
"
 The end result will be that the UnrealEngine hangs non-responsive, and gazebo aborts starting the GUI ending up with 
[Err] [ConnectionManager.cc:121] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:78] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

However if I use the following start sequence:
(PX4) make px4_sitl gazebo_iris
Unreal (./Myproject.sh) 
(airsim) ./GazeboDrone

PX4 will launch gazebo that binds to port 4560 since this is the simulator in PX4 context and the port is used for Steplock I guess gazebo should be the one that is the tcp server for port 4560

Unreal starts and complains among other things
"2022.06.22-21.26.31:525][  0]LogTemp: Opening mavlink connection
Opening mavlink connection
[2022.06.22-21.26.31:525][  0]LogTemp: Disconnecting mavlink vehicle
Disconnecting mavlink vehicle
Waiting for mavlink vehicle...
Disconnecting mavlink vehicle
Waiting for TCP connection on port 4560, local IP 127.0.0.1
Accepting TCP socket failed, is another instance running?
TcpClientPort socket bind failed with error: 98"

When starting GazeboDrone and QGroundControl everything starts to work. Maybe someone can shed some light on what I got wrong.

#### Context details
Ubuntu 20.04 LTS, 5.13.0-52-generic #59~20.04.1-Ubuntu SMP Thu Jun 16 21:21:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Airsim 1.7.0
Python 3.8.10
Unreal 4.26.2

settings.json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md
",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockType": "SteppableClock",
  "PhysicsEngineName":"ExternalPhysicsEngine",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "LockStep": true,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlPortLocal": 14540,
      "ControlPortRemote": 14580,
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": false,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },
	"Barometer":{
           "SensorType": 1,
           "Enabled": true,
           "PressureFactorSigma": 0.0001825
        }
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        }
      },
      "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1,
                "LPE_LAT": 47.62062,
                "LPE_LON": -122.34934

      },
      "X": 2, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "OriginGeopoint": {
      "Latitude": 47.62062,
      "Longitude": -122.34934,
      "Altitude": 149
  },
  "SubWindows": [
    {"WindowID": 1, "ImageType": 0, "CameraName": "front_center_custom", "Visibl
e": false}
  ]
}

### Include details of what you already did to find answers
 #2955,  #2380, #3284, https://github.com/jonyMarino/AirSim/tree/external_dynamics_simulator_with_example,

jonyMarino, 2022-06-24:
Hi and welcome @firemount ! The external physics engine feature doesn't use the flight controllers that AirSim uses. That is because there should be a way to inform the FDM about the power signal sent to the rotors by the FC to make it possible to use, for example, PX4. Given that, you should use the settings for a SimpleFlight drone and connect PX4 to your gazebo drone.

firemount, 2022-06-25:
Hi and thanks, that pointer made things a lot more clear.